iotsound/iotsound #7: Request: Add Airplay.

ReubenBTalbott, 2019-07-13:
No idea if this is possible to add to Balena-sound but could Airplay be added?

chrisys, 2019-07-22:
Sure, this is something we can look at! Thanks for the feedback

AlexProgrammerDE, 2019-07-25:
Maybe this could be a good help: https://github.com/mikebrady/shairport-sync

chrisys, 2019-07-26:
There's also the service here already set up to run on balena which you could copy in to the project https://github.com/balena-io-projects/boombeastic/tree/master/airplay

chrisys, 2019-10-04:
@ReubenBTalbott I am testing support for this in this PR https://github.com/balena-io-projects/balena-sound/pull/32 - if you get a chance give it a try and let me know what you think!

ReubenBTalbott, 2019-10-05:
I just tried it and so far it works perfectly, thanks!

chrisys, 2019-10-07:
Thanks for testing it, I've now merged this to master 👍

commit 67323290 (2019-10-07): Merge pull request #32 from balena-io-projects/add-airplay

Add support for Airplay streaming fixes #7
closed 2019-10-07 by chrisys