iotsound/iotsound #9: feature request: usb sound card support for raspberry pi zero W

yn1v, 2019-07-16:
I tried a raspberry pi zero with balena-sound. It worked with the HDMI audio output without any configuration on my part. Just had the HMDI cable at boot time. I tried with an USB sound card, but it didn't work having it plugged at boot time. Alsa Mixer list the device, but it can not change or select the output hardware.
I would like to have some tips on how to create a configuration file like .asoundrc or /etc/asound.conf that allows me set up an usb sound card for balena-sound. At the moment there is no file related to this. For instance, having a second take on the project page giving some ideas. Maybe a working example.
I guess, it will nice if pools the audio output and tries the USB devices "automagically" ... but I do understand that it is difficult to provide support to an unknown hardware. It is difficult to guess what kind of USB sound card adapter somebody will use.
Thanks

masterjuggler, 2019-07-18:
I agree that there should be some simple method of selecting an output. I just put balena-sound on my rpi 3B and have a USB DAC I'd like to use. In the meantime, I created an /etc/asound.conf like you mentioned and just added the lines below, substituting "card 1" with the number you want listed via `aplay -l`.

Create the file with `nano /etc/asound.conf` or whatever text editor you like, and add the block below. Restart and the chosen card will be selected in alsamixer.

```
pcm.!default {
    type hw
    card 1
}

ctl.!default {
    type hw           
    card 1
}
```
This worked fine for me, though if you want to switch between cards when you have multiple connected you'll need to switch the number manually.

Credits: I took that config chunk from here: https://superuser.com/a/630048

janvda, 2019-07-18:
@masterjuggler  I guess you created the file `/etc/asound.conf` in a terminal window of the `bluetooth-audio` service but then when you restart that service are any changes to files not undone by the balena docker engine ?

zvin, 2019-07-18:
@janvda You can create an `asound.conf` file in [bluetooth-audio](https://github.com/balena-io-projects/balena-sound/tree/master/bluetooth-audio) and add a `COPY asound.conf /etc/` line [here](https://github.com/balena-io-projects/balena-sound/blob/master/bluetooth-audio/Dockerfile.template#L32)

masterjuggler, 2019-07-18:
@janvda Sorry, you're right, I created the file using superuser permissions while under bluetooth-audio, I just edited my previous post to reflect that. I've since restarted my rpi and the changes persisted.

yn1v, 2019-07-18:
> 
> Create the file in a bluetooth-audio terminal with `sudo nano /etc/asound.conf` or whatever text editor you like, and add the block below. Restart and the chosen card will be selected in alsamixer.
> 

This is great. The different examples that I saw were lacking of context, if they will be good (or bad) for the specific OS (or the docker file). I was unsure where to start, as there was no file to modify. I was unsure which file to create and where to save it. I will try this on the weekend and report back.
Thanks very much for your help.

yn1v, 2019-07-19:
It is working!!
I am really a noob to containers and deployment. So, for anybody else like me, this is what I did. 

1. I went to the terminal in the audio container (as opposed as main os) and run the "aplay -l", so I got my sound card number. 0 is bcm (I am assuming is HMDI). 1 is my sound card

1. Went to my computer, looking for my git folder looked for balena-sound and there looked for bluetooth-audio folder.

1. Created the asound.conf file as the example. No need to change the example, as my sound card number is 1 like in the example. Your millage may vary.

1. Edited Dockerfile.template to include `COPY asound.conf /etc/`

1. Sent the new code to the device. My app is music... so `balena push music` and waited for the unicorn

1. Wait for the update to be present on the device... it shows the update bar progress (go get a cup of coffee)

1. Finally reboot de device (not sure if this is the best way to start running the new configuration/code... at least I did not pulled the cord)

I think that, if somebody look for this ... Now there is enough information to make it work. Although is not a nice prose with charts and drawings. 

Thanks for your help.

chrisys, 2019-07-26:
@yn1v that's great! Well done on getting it working! Have you thought about creating a PR so we can add the functionality to the project?

jvolker, 2019-08-17:
@yn1v Thanks for documenting your process. I just have a few questions. Unfortunately, it doesn't work for me yet. I don't get an audio signal through my USB sound card.

> Created the asound.conf file as the example. No need to change the example, as my sound card number is 1 like in the example. Your millage may vary.

Is the example what @masterjuggler posted here? https://github.com/balena-io-projects/balena-sound/issues/9#issuecomment-512652662

> Edited Dockerfile.template to include COPY asound.conf /etc/

Where in the file did you exactly include this? I tried after `COPY sounds /usr/src/sounds` and before `# Setup udev rules - this lets us play the connect/disconnect sound`.

Actually … I just checked whether `asound.conf` ended up in /etc/ on the device using the terminal of the audio container. It is indeed there.

This is what `aplay -l` returns:

```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Crimson [SPL Crimson], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
I assume I should be able to use the example by @masterjuggler straight away as well, right?

So what else could be the problem?

Thanks.

jvolker, 2019-09-02:
Are there any tests I could run or logfiles to get some more information / debug regarding this issue?

Thanks.

yn1v, 2019-09-04:
> Is the example what @masterjuggler posted here? [#9 (comment)](https://github.com/balena-io-projects/balena-sound/issues/9#issuecomment-512652662)
> 

Yes, that is the example. The asound.conf file contains the following lines:

```
pcm.!default {
    type hw
    card 1
}

ctl.!default {
    type hw           
    card 1
}
```

> > Edited Dockerfile.template to include COPY asound.conf /etc/
> 
> Where exactly in the file did you include this? I tried after `COPY sounds /usr/src/sounds` and before `# Setup udev rules - this lets us play the connect/disconnect sound`.
> 

Yes, exactly there. 

> Actually … I just checked whether `asound.conf` ended up in /etc/ on the device using the terminal of the audio container. It is indeed there.
>
> This is what `aplay -l` returns:
>
> ```
> **** List of [...]

I guess so. The "card 0" is BCM, as far as I understand is the sound output of the HDMI. And your usb card is "card 1"

> So what else could be the problem?

Sorry about the long time that took to reply. I have no idea, I am a noob. It was a challenge for me to figure it out as I am not familiar with command line sound cards and also it was my first experiment with containers.

The result that I got is that the blue-tooth audio is lagging because the raspberry pi zero is overloaded with the container. Not sure what will be the correct term, the audio stops playing for a couple of seconds and then resumes. Gaps? Skips? ... it is beyond quality (noise, volume, delay to start). Is not a continuous playback of the audio. So much work and for nothing.

janvda, 2019-09-04:
@yn1v , I am sorry to hear you have these problems.
The good news is that you got the USB sound card "configured".

I think that the current problem you are facing should also be solvable.
Maybe it is worth creating a separate issue for that in this repository.

Some next steps regarding troubleshooting this issue:

- During bad playback : you can run `top` command in a terminal window of your container in the BalenaCloud dashboard.  This let you see how busy your container is.
- I would also try to play an audio file you have downloaded in your container using the `aplay` command.  E.g. you can download a sample audio file by following command `wget https://www.kozco.com/tech/LRMonoPhase4.wav` (see also https://github.com/janvda/balena-pulseaudio/blob/master/pulseaudio-client/Dockerfile.template )  If this audio file is played properly then this indicates that it is the bluetooth connection that is causing the issue.

jvolker, 2019-09-06:
Thanks for your reply @yn1v. 

I thought of some ways to debug my issue. I don't have the time or hardware to do it right now but will hopefully get back to it later.
1. Check if the device is playing back via HDMI (with HDMI display with speakers or tv).
2. Play an audio file with `aplay` like @janvda described in https://github.com/balena-io-projects/balena-sound/issues/9#issuecomment-527952061 to see if the issue is related to bluetooth or the alsa soundcard configuration.
3. Read https://alsa-project.org/wiki/Asoundrc to get a better understanding of how the config file works and to see if there are any other options.

Hope this is helpful to either me or someone else in the future.

TomasValenta, 2019-10-21:
My soundcard is the **Bose USB Audio**, detected by `aplay -l` as device 1, but the above how-to didn't work for me (tried using bluetooth audio and spotify connect) :(

From previous experience with **raspotify** configuration on Raspbian, I tried to simply disable the onboard audio card and is working fine for me! :) The only change I had to make was changing the **DT param audio** from on to **off** in the Device configuration tab and restart the device ;)

[image]

It may also help someone.

AlexProgrammerDE, 2020-05-02:
We included in the new documentation all tips from this issue. You can check it out here: https://landr-balenalabs-repo-balena-sound.netlify.app/docs/troubleshooting#no-audio . Closing

closed 2020-05-02 by AlexProgrammerDE

pjgoodall, 2020-07-18:
I have an external amplifier with a USB DAC input. The outputs of `alsa -l`  and `lsusb ` are below.

i forked balena-sound, and created the `asound.conf` file and added the `COPY asound.conf /etc/` to `bluetooth-audio/Dockerfile.template`

I pushed the resulting configuration using 'balena push' and it arrived on my device - no sound.
Following basic instructions on the [Asoundrc](https://alsa-project.org/wiki/Asoundrc#The_naming_of_PCM_devices) I successfully played high-quality sound using:

``` bash
aplay -D hw:1,0 LRMonoPhase4.wav
```
So nothing wrong with my hardware.

I've done everything I can work out. Can someone advise please.

Thanks
--Peter G

``` bash
% cat /proc/cpuinfo | grep Model
Model           : Raspberry Pi 2 Model B Rev 1.1
```

``` bash
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: CODEC [USB Audio CODEC], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

# lsusb
Bus 001 Device 008: ID 08bb:2902 Texas Instruments PCM2902 Audio Codec
Bus 001 Device 004: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

Dricc123, 2020-12-30:
Dear All, what is the latest status? I see above that issues have been closed? Does Balena-Sound support Rpi Zero through usb now? I'm a newbie. Just tried Volumio on Rpi Zero and usb soundcart is not detected.