iotsound/iotsound #10: Bluetooth not discoverable in raspberry pi 3 model B+

mvenilton, 2019-07-17:
Works like a charm in raspberry pi 3 model B, but in model B+ the bluetooth is not discoverable in phone or pc.

Nytrouz-official, 2019-07-18:
I just played a little bit with the lock overdrive button dont know what it does but now my mobilephone is conected
(pi3 B+ same issue)

chrisys, 2019-07-22:
@mvenilton interesting! I just deployed to a new 3B+ and see the same thing.

If you use the dashboard terminal and run `bluetoothctl` and then `show` you can see the discoverable status. You can also force it on by running `discoverable on`. I will take a look to see if I can figure out why this is happening.

chrisys, 2019-07-24:
You should find this now works ok with v0.0.10 - can you give it a try and reopen this issue if not?

closed 2019-07-24 by chrisys

mvenilton, 2019-07-26:
> 
> 
> You should find this now works ok with v0.0.10 - can you give it a try and reopen this issue if not?

Thanks @chrisys This is working!

Nytrouz-official, 2019-08-01:
how do I update to the new version do I just push it ore do I have to delete the old one?

chrisys, 2019-08-01:
@Nytrouz-official you can download the new version from GitHub and push it again to your device, you don't have to flash the SD card or delete anything. The build system will recognise that it's only a small change.

Nytrouz-official, 2019-08-01:
ok thanks