iotsound/iotsound #15: Error on startup: /usr/bin/bluealsa: Couldn't acquire D-Bus name: org.bluealsa

pwt, 2019-07-31:
I had this working fine until the last couple of days. Now I get the following error when the container starts:

```
Setting volume to 100%
Restarting bluetooth service
Bluetooth agent registered
/usr/bin/bluealsa: Couldn't acquire D-Bus name: org.bluealsa
Device is discoverable as "balenaSound"
/usr/bin/bluealsa-aplay: Couldn't get BlueALSA PCM list: The name org.bluealsa was not provided by any .service files
```

There was a Balena image rebuild, but no relevant code changes as far as I'm aware. Deploying an older release avoids the error.

Any ideas?

pwt, 2019-07-31:
I've verified that this is happening with the current upstream release (https://github.com/balena-io-projects/balena-sound/commit/521cca4fcc9af35a8f3f1db3d2e211e81b12d77f) with no code changes.

I'm running on a Raspberry Pi Model 3 B+, with the current Balena Host OS (balenaOS 2.38.0+rev1) and Supervisor (9.15.7) versions. The docker-compose.yml file has the required entries.

pwt, 2019-07-31:
As a final check, I deleted the device and started from a freshly imaged memory card. No change in behaviour.

Mishco, 2019-08-01:
Show me the same error in logs.

* Raspberry Pi Model 3 B+  
* balenaOS 2.38.0+rev1
* Supervisor 9.15.7 
* v0.0.10

bderry71, 2019-08-01:
I'm new to Balena.io and this is my first project to load on my Pi with their BalenaOS. I'm glad I'm not the only one seeing this error. 

Raspberry Pi 3 B+
balenaOS 2.38.0+rev1
Supervisor 9.15.7
balena-sound v0.0.10

pwt, 2019-08-01:
As I suspected, the problem is due to the switch to Buster from Stretch. If you use Stretch, everything works as before. The first line in Dockerfile.template needs to be changed as follows:

```
-FROM balenalib/%%BALENA_MACHINE_NAME%%:buster
+FROM balenalib/%%BALENA_MACHINE_NAME%%:stretch
```

Probably best to revert back to Stretch while the Buster bugs/changes are worked out. I'm hitting Buster issues in a number of different areas.

I'll create a PR.

txgruppi, 2019-08-01:
Thanks @pwt, it is working fine now.

closed 2019-08-01 by chrisys