iotsound/iotsound #17: Cannot play audio on linux PC

BarryAlvord, 2019-08-03:
After checking my log on Balena cloud, Here what it shown:

03.08.19 14:06:35 (+0700) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1
03.08.19 14:06:35 (+0700) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1
03.08.19 14:06:35 (+0700) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1
03.08.19 14:06:35 (+0700) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1
03.08.19 14:06:35 (+0700) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1
03.08.19 14:06:35 (+0700) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1
03.08.19 14:06:35 (+0700) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1
03.08.19 14:06:35 (+0700) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1

it shown this for entire of the log file

Here its my issuse report.Thanks for reading this

AlexProgrammerDE, 2019-10-17:
Try it with another balenaos version or anothe bluetooth adapter.

tmigone, 2020-01-10:
Hey @BarryAlvord, in case you didn't figure it out, please try with the latest balenaOS version and let us know how it goes. Cheers.

AlexProgrammerDE, 2020-01-11:
@tmigone i think thia account is no longer active.
btw: What is with linux PC ment? balenaSound is for the Raspberry Pi. No wonder that this is not working.

chrisys, 2020-01-27:
Closing due to lack of activity = @BarryAlvord if you have some more info and a specific issue for us to look at please feel free to reopen.

closed 2020-01-27 by chrisys

stalkerGH, 2020-02-26:
I can add something more to this topic. I face exactly the same problem. I tried it on different setups:

1) RPi 2 with cheap Bluetooth dongle and connected by cable Ethernet - it is detected by balenaSound, BT LED blinks, no sound from speaker or headphones connected to Pi's audio out. I tried two (latest and previous) balenaOS versions,
2) RPi 3 (32 bit version) - WiFi and BT built-in, latest balenaOS version.

All three setups - no sound, only "Unsupported RTP payload type: 1" error. What should I do?

VincentChalnot, 2020-06-16:
Same problem here, I can bind my linux laptop to the RPi 3A+ but no sound is played with the very same error.
I've never experienced any problem with this machine before, I think it might be related to the bluetooth "codec" used on the PC : High Fidelity A2DP Sink ?

VincentChalnot, 2020-06-16:
It's definitely linked to this issue: https://github.com/Arkq/bluez-alsa/issues/13