iotsound/iotsound #19: Unable to locate package bluealsa

danielsteiner, 2019-08-08:
I just tried to push this project to my raspberry but I get an error when it tries to download bluealsa. 

```
[Info]             Dashboard link: https://dashboard.balena-cloud.com/apps/1499654/devices
[Info]             Building on arm01
[Info]             Pulling previous images for caching purposes...
[Success]          Successfully pulled cache images
[bluetooth-audio]  Step 1/13 : FROM balenalib/raspberrypi3-64:stretch
[bluetooth-audio]   ---> 2c6daafcf5bf
[bluetooth-audio]  Step 2/13 : ENV DBUS_SYSTEM_BUS_ADDRESS unix:path=/host/run/dbus/system_bus_socket
[bluetooth-audio]  Using cache
[bluetooth-audio]   ---> c9e969ef66c9
[bluetooth-audio]  Step 3/13 : ENV UDEV 1
[bluetooth-audio]  Using cache
[bluetooth-audio]   ---> c3176c9feb77
[bluetooth-audio]  Step 4/13 : RUN install_packages   alsa-utils   bluealsa   bluez   python-gobject   python-dbus   mplayer
[bluetooth-audio]   ---> Running in 60fe2abc1a76
[bluetooth-audio]  Reading package lists...
[bluetooth-audio]  Building dependency tree...
[bluetooth-audio]  Reading state information...
[bluetooth-audio]  E: Unable to locate package bluealsa
[bluetooth-audio]
[bluetooth-audio]  apt failed, retrying
[bluetooth-audio]  Reading package lists...
[bluetooth-audio]  Building dependency tree...
[bluetooth-audio]  Reading state information...
[bluetooth-audio]  E: Unable to locate package bluealsa
[bluetooth-audio]
[bluetooth-audio]  apt failed, retrying
[bluetooth-audio]  Reading package lists...
[bluetooth-audio]  Building dependency tree...
[bluetooth-audio]  Reading state information...
[bluetooth-audio]  E: Unable to locate package bluealsa
[bluetooth-audio]
[bluetooth-audio]  Removing intermediate container 60fe2abc1a76
[Info]             Uploading images
[bluetooth-audio]  The command '/bin/sh -c install_packages   alsa-utils   bluealsa   bluez   python-gobject   python-dbus   mplayer' returned a non-zero code: 100
[Success]          Successfully uploaded images
[Error]            Some services failed to build:
[Error]              Service: bluetooth-audio
[Error]                Error: The command '/bin/sh -c install_packages   alsa-utils   bluealsa   bluez   python-gobject   python-dbus   mplayer' returned a non-zero code: 100
[Error]            Not deploying release.
```

chrisys, 2019-08-15:
@danielsteiner I see you're using the 64-bit Raspberry Pi 3 image, could you try with the regular Raspberry Pi 3 image?

Jamdoog, 2019-08-25:
Where do I find the regular Raspberry Pi 3 image? I cannot find it on Balena dashboard...

chrisys, 2019-08-26:
@Jamdoog I replied to your forum post but... You can’t (currently) mix 32 and 64bit Raspberry Pi devices within the same application, so the fact you’ve set the application default as 64bit Raspberry Pi means that this is the only one of the two devices that you can select on the add new device screen. If you create a new application using Raspberry Pi 3 as the default then you’ll be able to add one.

closed 2019-08-29 by chrisys

xginn8, 2019-09-02:
@chrisys maybe worth noting this in the blog post/README?

danawoodman, 2019-10-01:
There are no options for RPi 4 non-64bit in the new application modal: https://cl.ly/a1490cab9abb/Screen%20Shot%202019-10-01%20at%201.24.11%20PM.png

chrisys, 2019-10-01:
@danawoodman Raspberry Pi 4 is 64-bit only.

danawoodman, 2019-10-01:
@chrisys well the above error occurs when I follow the tutorial and run `balena push <project>` with:

```
E: Unable to locate package bluealsa
```

chrisys, 2019-10-01:
@danawoodman this project doesn't support 64-bit OS (and hence Raspberry Pi 4)... yet. There's an issue open here: https://github.com/balena-io-projects/balena-sound/issues/25

danawoodman, 2019-10-01:
Ah damn, ok, wish I would have know that before I started 😢  I'll try with a Pi 3

chrisys, 2019-10-01:
@danawoodman I'm taking a look at running it on a Pi 4 right now, but yeah if you try a Pi 3 (not 64-bit), you should be set!

chrisys, 2019-10-01:
@danawoodman if you edit the first line of `bluetooth-audio/Dockerfile.template` to read `FROM balenalib/raspberrypi3:stretch`, it might work on your Pi 4.

chrisys, 2019-10-01:
@danawoodman sorry for the multiple mentions but wanted to let you know incase you were still working on it, I just opened a PR with a simple fix to get it running on Pi 4 (essentially forcing a 32-bit image). It likely won't be merged until tomorrow but the branch is here if you wanted to try it: https://github.com/balena-io-projects/balena-sound/tree/support-pi4

danawoodman, 2019-10-03:
@chrisys thanks for the fast action! I will give the new Pi 4 support a test!

I tried on a Pi 3 and could never pair my devices, so hopefully the Pi 4 support does the trick!

chrisys, 2019-10-03:
@danawoodman that's unusual, the Pi 3 is usually pretty solid. If you still have trouble with the Pi 4 open an issue and let me know what devices you're trying and I'll take a look.

danawoodman, 2019-10-07:
@chrisys when I get some free time again I will dive back in and try it. Before it would just boot me out via SSH even after making sure the item was not in known_hosts

danawoodman, 2019-10-07:
I was able to get RPi 4 working on the latest branch! At first it was acting strange but I rebooted and it seems to work pretty well! I'll report back if I find any issues

chrisys, 2019-10-08:
@danawoodman great stuff, thanks for the feedback!

danawoodman, 2019-10-12:
So far the Pi 4 has been working solidly 👍

chrisys, 2019-10-12:
@danawoodman great to hear, cheers!