iotsound/iotsound #21: Bluetooth doesn't connect automatically when the service starts

iRunn3r, 2019-08-21:
Upon startup, Bluetooth doesn't connect automatically with paired devices on startup. This is very annoying when using BalenaSound in the car as you need to navigate to phone's Bluetooth settings and locate the Bluetooth connection entry.

Steps to reproduce the issue:
1) Power on the Raspberry Pi with BalenaSound installed.
2) Connect a phone to BalenaSound Bluetooth service.
3) Restart the Raspberry Pi (unplug and plug back in).

Expected: Bluetooth automatically re-connects.
Actual: Bluetooth doesn't re-connect automatically, manual connection is needed.

Environment:
iPhone XS Max iOS 13.0 (haven't checked Android).
Raspberry Pi 3B
balenaOS 2.38.0+rev1
bluetooth-audio rev f623957

mcmuffindk, 2019-09-06:
I am having kinda the same issue

When you unplug and plug the Pi back in i have to remove it from "My Devices" and then connect to it from scratch or it will error on connection

Enviroment:
iPhone XS iOS 12.4
Raspberry Pi Zero WH
balenaOS 2.32.0+rev1
Balena-Sound master v0.0.11

AlexProgrammerDE, 2019-10-16:
It seems to me hard to bring this into balena-sound. But when you give me the MAC adress  of your mobile can i write a script which connects you automatically per bluetoothctl to your RPI 3. You need then to put the updated file into your copy of balena-sound and then build it again. If you dont know how you get rhe MAC adress i can try to give you instructions, but please try it first yourself.

tmigone, 2019-11-18:
Hello, I've implemented reconnection to last known device on PR #57 . If you guys get a chance give it a try and let me know what you think.

tmigone, 2019-11-20:
Closing issue as PR #57 was merged into master. Please let me know if you have issues with this. Thanks!

closed 2019-11-20 by tmigone