iotsound/iotsound #23: Cant play any music when connected with iPad

VoxelTek, 2019-08-29:
Whenever I boot the RPi and i see the log on the dashboard, this comes up:
`amixer: Unable to find simple control 'PCM',0`
i ignore it, as i do not think it will affect anything. I connect my iPad(or whatever device i am using) to the bluetooth and start playing some music. when i hear nothing, i go back to the log and see a lot of errors in there. the errors are:
```
/usr/bin/bluealsa-aplay: Couldn't open PCM: No such device

ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave

/usr/bin/bluealsa: SBC decoding error: No such file or directory

/usr/bin/bluealsa: Missing RTP packet: 4157 != 4156

```
 however, when i press pause on my ipad, the errors stop spitting out. all this time, there is absolutely no sound coming through the speakers, which i have tested. i really want to be able to use bluetooth with these speakers. please resolve this issue immediately.

From,
VoxelTek

AlexProgrammerDE, 2019-10-06:
Which Pi are you using?

VoxelTek, 2019-10-06:
I am using a Raspberry Pi 3B

AlexProgrammerDE, 2019-10-07:
Have you tried reinstalling?

VoxelTek, 2019-10-13:
Yes, multiple times. I have had the same error every time.

AlexProgrammerDE, 2019-10-13:
Are you using the on board jack or a external sound card (USB, GPIO, etc.)?

VoxelTek, 2019-10-13:
The on board jack.

AlexProgrammerDE, 2019-10-13:
Try it please with:

`apt-get update && apt-get install --reinstall alsa-utils`

(In the bluetooth-audio container which is aviable at the cloud dashboard)

VoxelTek, 2019-10-13:
Ok thanks

AlexProgrammerDE, 2019-10-17:
Try it please with balenaOS 2.38.0+rev1 production this was a solution in #47 which had the same errors.

AlexProgrammerDE, 2019-10-31:
@VoxelTek Did it help?

VoxelTek, 2019-10-31:
Unfortunately, I have been too busy to try any of these solutions. I should be able to try them in 2 days time.

AlexProgrammerDE, 2019-10-31:
Ok

tmigone, 2020-01-10:
Hey @VoxelTek, whenever you have time, please feel free to try again with the latest version of balenaOS as Alex suggested. Cheers!

willenglishiv, 2020-01-10:
I have a fresh install of this on a raspberry pi 3B and I'm experiencing the same issue.  I connected my galaxy phone through spotify connect and bluetooth and nothing comes out of the audio port.  same error logs.

AlexProgrammerDE, 2020-01-11:
@willenglishiv Can you give me the health test reports please?  You can get it at the bottom left of the dashboard.

chrisys, 2020-01-27:
@willenglishiv  @VoxelTek can you confirm you are not using the 64-bit OS and give it a try with the latest version. If the issue still persists after this please feel free to reopen the ticket.

closed 2020-01-27 by chrisys