iotsound/iotsound #24: feature request: Could we include multiple connections

kitovh, 2019-09-04:
I'm trying to figure out if there would be a way to allow multiple bluetooth devices to connect at the same time. When playing music from all devices, it would just cue whatever comes next regardless of who requested it.

Best,

chrisys, 2019-10-04:
@kitovh that wouldn't be possible with the bluetooth audio setup, as it's streaming literally just the audio over the bluetooth, it has no concept of what that audio is or when tracks begin and end. Perhaps it would be possible via another protocol though?

AlexProgrammerDE, 2019-10-17:
In https://github.com/Arkq/bluez-alsa/issues/70 was this mentioned.

AlexProgrammerDE, 2020-04-10:
@chrisys @kitovh i have just tested it on my RPI 4 1GB. My two connected android phones were able to play music. But if both devices play at the same time i get RTP errors. Maybe is this caused by the firmware, which cant handle so many packets. As long as only one device plays music is this ok. I will close for now. Feel free to reopen if you have still problems with it.

closed 2020-04-10 by AlexProgrammerDE