iotsound/iotsound #26: Performance on Raspberry Pi Zero good enough?

jvolker, 2019-09-06:
As described by @yn1v in https://github.com/balena-io-projects/balena-sound/issues/9#issuecomment-527732187 the performance by the Pi Zero doesn't seem to be good enough for this application. 

Can someone with a Pi Zero please share their experiences?
Is the performance on all other versions of the Pi okay?

@janvda described ways to debug performance issues on the Pi Zero:
https://github.com/balena-io-projects/balena-sound/issues/9#issuecomment-527952061

Thanks.

xXnijamanXx, 2019-09-21:
After seting up the pi zero WH and connection to the bluetooth I seem to have the same issue as @yn1v the Pi will skip, quickly play a section at almost double speed.
I have not yet tried to extensively test the issue but just from these preliminary signs especially the audio quickly playing to "catch up" indicates to me the Pi being overwhelmed then quickly spitting out the audio to catch up. Could also be a problem with the DAC itself but I'm less suspicious of it. 

Will update with any results from my testing.

xXnijamanXx, 2019-09-21:
**Update**
It seems to be a Bluetooth Issue after using the  `top` command in Balena the usage would never rise over 30%.

I also noticed that testing distances with my phone from the Pi Zero WH that as I got closer to the device the sound became much more stable with less "skips" and "catch-ups". If the issue is the Bluetooth ability of the Zero perhaps there is an external Bluetooth adapter that could be fitted for higher quality Bluetooth connection.

Also given that the Pi Zero only uses "Bluetooth Low Energy (BLE)" [https://en.wikipedia.org/wiki/Bluetooth_Low_Energy#Compatibility](url) looking at the specs for the BLE it's "data rate over air" is much lower than the "Bluetooth Basic Rate/Enhanced Data Rate technology".  On that wiki article and on slide 3 of this power point from the IEEE [https://californiaconsultants.org/wp-content/uploads/2014/05/CNSV-1205-Decuir.pdf](url) it is said that the Low Energy Bluetooth is not suitable for streaming audio video and other such things mainly useful for sending states and things as such.

For me it seems that either you would have to send very compressed data to the Zero compromising the sound quality drastically or there may be external Bluetooth adapters that could be used inside of the on-board chip.

janvda, 2019-09-21:
The pi zero WH also seems to support Bluetooth 4.1 - so not only BLE - which should be suited for audio streaming.

chrisys, 2019-10-04:
I've started noticing issues with this after upgrading my iPhone to iOS 13. Prior to that, and from other devices, it's been working fine on a Pi Zero for me. I'll try to spend some time looking into it soon!

jtagcat, 2020-04-23:
Raspberry Pi Model B Rev 2, every time the audio skips, I get:
`23.04.20 11:01:23 (+0000)  spotify  ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred`.

Removing bluetooth seems to help a lot (there is no bt on my device), lowering bitrate with `--bitrate 320` in spotify/start.sh to the [documented](https://github.com/librespot-org/librespot/wiki/Options) 160 seems to resolve all issues.

I haven't bothered to dig in to alsa, as the above solutions work. If you decide to dig in: [[1]](https://stackoverflow.com/questions/14601666/alsa-ways-to-prevent-underrun-for-speaker) [[2]](https://bbs.archlinux.org/viewtopic.php?id=185736)

georgiats, 2020-08-17:
comment

champagj, 2020-08-27:
I have tested deploying balenasound on a Rpi0 using the one-click depploy option.
I did experience some underrun errors (cutting sound) when listening to Spotify using SpotifyConnect (libespot).
I updated the start.sh in spotify folder to change the bitrate from 320 to 160 as follow (bottom of the file)

```
# Start raspotify
if [[ $DISABLE_VOLUME_NORMALISATION == "1" ]]; then
  exec /usr/bin/librespot --name "$DEVICE_NAME" $SPOTIFY_BACKEND --bitrate 160 --cache /var/cache/raspotify --linear-volume --initial-volume=$SYSTEM_OUTPUT_VOLUME $SPOTIFY_CREDENTIALS $SPOTIFY_DEVICE
else
  exec /usr/bin/librespot --name "$DEVICE_NAME" $SPOTIFY_BACKEND --bitrate 160 --cache /var/cache/raspotify --enable-volume-normalisation --linear-volume --initial-volume=$SYSTEM_OUTPUT_VOLUME $SPOTIFY_CREDENTIALS $SPOTIFY_DEVICE
fi
```
After I pushed the update via balena cli the issue of underrun appears to have stopped.

tmigone, 2022-04-05:
Quick update regarding this topic.

Raspberry Pi Zero/1's are 7 and 10 years old at this point. We want to keep supporting these boards because a lot of people have them and the project for the most part runs fine on them, but our focus will be on newer boards: Pi 3, 4 and Zero 2. 

This means going forward its possible new features might not get added for Pi Zero/1. We haven't decided yet if we will create a lightweight version of balenaSound which would run on low end devices or if we would keep everything on the same project and use an architecture blacklist like we do to disable multiroom server on these devices.

As a final note I just wanted to clarify that as of `v3.8.9` the project runs fine on Raspberry Pi Zero and Raspberry Pi 1. That said here are a few suggestions you can try if you find your device is struggling to properly run balenaSound:
- make sure you are using a good quality power supply
- make sure you are using a good quality SD card, we recommend SanDisk Extreme Pro
- make sure your device is not too far away from the WiFi router

Most of the issues we've seen were Pi Zero's struggle to run balenaSound are related to the suggestions described above, if you are still having trouble feel free to contact us at https://forums.balena.io/ for a more in depth troubleshooting.