iotsound/iotsound #27: Can't connect to device after powerloss

mcmuffindk, 2019-09-11:
When you unplug and plug the Pi back in i have to remove it from "My Devices" and then connect to it from scratch or it will error on connection

Enviroment:
iPhone XS iOS 12.4
Raspberry Pi Zero WH
balenaOS 2.32.0+rev1
Balena-Sound master v0.0.11

chrisys, 2019-09-19:
We’ve seen this before, and it has been resolved in balenaOS versions 2.38 and higher. I’m guessing as you’re still seeing it now that you’re running a Raspberry Pi Zero as the current version of balenaOS on the dashboard is 2.32 which exhibits this problem; other Pis have more recent OS versions available.

You can work around this by downloading the latest OS for the Pi Zero from staging: https://dashboard.balena-staging.com/login If you combine a download of config.json from the normal dashboard with the OS download from staging, you should be able to get it working. You can flash the staging OS to your SD card, then plug it back into your computer and replace the config.json file within the resin-boot partition with the one downloaded from your app on the production dashboard.

tmigone, 2020-01-10:
Closing this issue in favor of Chris' solution: update balenaOS to version 2.38 or higher.

closed 2020-01-10 by tmigone