iotsound/iotsound #30: Installing Without Balena

capriciousduck, 2019-10-02:
Hi, a day ago I installed stretch on my Pi 3. I want to try balena sound. Is it possible to deploy this app without using balena OS?

chrisys, 2019-10-02:
@capriciousduck of course! The [Dockerfile](https://github.com/balena-io-projects/balena-sound/blob/master/bluetooth-audio/Dockerfile.template) essentially forms a list of instructions for you to carry out manually.

closed 2019-10-03 by chrisys

capriciousduck, 2019-10-03:
I build a new docker images using the file your referred in the above comment. I am getting the following error. Also I set host networking mode and also spinned the container up in privileged mode.

Failed to open connection to "system" message bus: Failed to connect to socket /host/run/dbus/system_bus_socket: No such file or directory
Setting volume to 100%
Restarting bluetooth service
Failed to open connection to "system" message bus: Failed to connect to socket /host/run/dbus/system_bus_socket: No such file or directory
Setting volume to 100%
Restarting bluetooth service

capriciousduck, 2019-10-03:
Got it to work!

Installed python-dbus on host and mounted /var/run/dbus/system_bus_socket to container.

chrisys, 2019-10-03:
@capriciousduck nice work!

jakubladra, 2020-04-11:
I followed ghost's approach and then ran `run docker run --privileged balena-audio -v /var/run/dbus/system_bus_socket:/host/run/dbus/system_bus_socket` (balena-audio is my name for the built docker image). It seems that I'm failing on running start.sh script:

```
mount: special device /dev/console does not exist
umount: /dev: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
/usr/bin/entry.sh: line 59: exec: : not found
```

Do I have to pass any environment variables that would be used in the entry script?

Is it suppose to make the Raspberry Pi discoverable over bluetooth once the script runs well?

Thanks!

b-a0, 2020-12-29:
@jakubladra Did you ever get this to work? Do you maybe have some pointers or a Dockerfile to share?