iotsound/iotsound #33: Alsa Settings Reset

capriciousduck, 2019-10-05:
Hi Chris, I am really enjoying balena sound. I installed it using the dockerfile(no balena OS). I also switched to ethernet as I thought it will improve my LAN speeds as I am using the Pi as a small NAS and will improve Bluetooth functionality(besides Cypress fix). Everything is working great but the volume setting is resetting.

I am increasing the volume through alsamixer and storing the settings using alsactl, but I still see the volume reset to 36%. Even when I increase the audio in my phone, I still see no change.

Thank you for your wonderful project! I have searched for many days for something like this. I really like the docker thing. Spin up and spin down!

AlexProgrammerDE, 2019-10-07:
@capriciousduck You can use the function added by chrisys with which you can set the system output of the bluetooth music container. Add to the template:

> ENV SYSTEM_OUTPUT_VOLUME 100

And replace 100 with your wished Volume.
Also you need to delete the following line from start.sh:

>

AlexProgrammerDE, 2019-10-11:
@capriciousduck Has it worked?

capriciousduck, 2019-10-11:
No. It is still resetting.

capriciousduck, 2019-10-31:
I am using Dietpi with Docker installed.

AlexProgrammerDE, 2019-10-31:
In https://dietpi.com/phpbb/viewtopic.php?t=1508 was the solution: sudo

capriciousduck, 2019-10-31:
I am a root user. I am using the latest image of dietpi. It only has root
user by default. So all the commands I run, will run as root.

AlexProgrammerDE, 2019-10-31:
How about this: https://community.roonlabs.com/t/save-last-volume-level-with-dietpi/47635

capriciousduck, 2019-10-31:
Now I see that the volume is resetting every time I connect and disconnect bluetooth. After connecting to my bluetooth device, I always have less volume output in Pi. So I set to desired volume using alsactl. After I play music  for a while and disconnect and then connect again at some other time the volume is resetting.

I set the volume env variable too. But that doesn't work.

AlexProgrammerDE, 2019-10-31:
How about changing bluetooth-udev everytime you connect and disconnect your specific commands are running.

capriciousduck, 2019-10-31:
I don't know much about Bluetooth config. Can you please tell me exactly
what I should do?

AlexProgrammerDE, 2019-10-31:
So the bluetooth-udev file is a bash script, which runs specific code in a specific case appears. You will see it when you edit it. :smile_cat:

AlexProgrammerDE, 2019-10-31:
There can you add at the end the from you used commands. I dont know if it works. Just try.

AlexProgrammerDE, 2019-11-02:
And did it work?

tmigone, 2020-01-22:
Possibly related to #78

AlexProgrammerDE, 2020-04-01:
Closing, because the account got deleted and this issue got possibly resolved by #129 .

closed 2020-04-01 by AlexProgrammerDE