iotsound/iotsound #42: Balena shutdown

bloppe, 2019-10-08:
Hi, 

I've just completed my first Balena project, the Balenasound raspberry pi build. It works great, but I am not sure how to remotely shut my pi down without just turning it off at the wall. Help please

AlexProgrammerDE, 2019-10-08:
[image]
[image]
[image]
[image]
[image]
[image]

AlexProgrammerDE, 2019-10-08:
Go to balena then login then click on the blue device name then click on the arrow and then click shutdown.

chrisys, 2019-10-08:
@AlexProgrammerDE thanks for replying here! 👍 As mentioned above, the shutdown option is on the dashboard in the device screen under the dropdown.

[image]

closed 2019-10-10 by chrisys