iotsound/iotsound #47: connection is buffering

mabli, 2019-10-14:
Hello, 

i just installed the balena sound project on y raspi and was very happy that i got it running. :-)
I use it with a raspi 3 and a justboom amp hat soundcard. I can connect via bluetooth, but the song is buffering or the connection is not stable. i don't know. in the logs there are these messages in multiple variations.  .

14.10.19 10:35:29 (+0200) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1
14.10.19 10:36:43 (+0200) <bluetooth-audio> /usr/bin/bluealsa: Missing RTP packet: 1227 != 1222
14.10.19 10:36:51 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such file or directory
14.10.19 10:37:24 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such process

i use balenaOS 2.43.0+rev1 in the developement version. For the use with the justboom amp i created a custom variable BALENA_HOST_CONFIG_dtoverlay with the value "justboom-amp". The entry of the value was pretty much a guess after in the manual was mentioned only the case with hiffierry-dac. Could this cause the problem?

Greetings Max

AlexProgrammerDE, 2019-10-15:
https://www.justboom.co/software/configure-justboom-with-raspbian/ has some maybe for you usefull information.

AlexProgrammerDE, 2019-10-15:
The page has this info:

dtoverlay=i2s-mmap dtoverlay=justboom-dac

I guess you tried it with justboom-amp but the page uses for the amp justboom-dac

mabli, 2019-10-15:
The change to "justboom-dac" seems to have been the answer to the logs. After the change there are no more logs till now. Still i didn't enter the "is2-mmpa" part. What does it?

should i make the entry in that form in the value of the Costum Variable BALENA_HOST_CONFIG_dtoverlay
BALENA_HOST_CONFIG_dtoverlay    "justboom-dac", "i2s-mmpa"

But the stagnant connection is still there. The error appears with an android tablet and a windows laptop. with an ubuntu laptop there is a connection possible but no sound output.. I still try to figure out why.
Is it possible, that the bluetooth module from the raspi 3 is too weak or defect?

AlexProgrammerDE, 2019-10-15:
RTP means Real-time Transport Protocol (RTP). You can read more about it in Wikipedia: https://en.wikipedia.org/wiki/Real-time_Transport_Protocol .  And Bluealsa says it is Unsupported.
It is maybe a dtparam error. The web page says you need to set

dtparam=audio=off

Maybe can you then play sounds. Good Luck-
PS: BALENA_HOST_CONFIG_dtoverlay "justboom-dac", "i2s-mmpa" is should be right (I saw something like this before). I am not very informed in this.

mabli, 2019-10-15:
Applying boot config: {"dtoverlay":["i2s-mmpa","justboom-dac"],"disable_splash":"1","dtparam":["i2c_arm=on","spi=on","audio=off"],"enable_uart":"1","gpu_mem":"16"}

15.10.19 16:41:20 (+0200)  bluetooth-audio  Setting output volume to 100%
15.10.19 16:41:20 (+0200)  bluetooth-audio  Connection notify volume is 75%
15.10.19 16:41:21 (+0200)  bluetooth-audio  amixer: Unable to find simple control 'PCM',0
15.10.19 16:41:21 (+0200)  bluetooth-audio  
15.10.19 16:41:21 (+0200)  bluetooth-audio  Restarting bluetooth service
15.10.19 18:10:22 (+0200)  bluetooth-audio  Bluetooth agent registered
15.10.19 18:10:25 (+0200)  bluetooth-audio  Device is discoverable as "balenaSound 818b"

i followed your suggestions. but the stagnant connection is persistent. while the tablet still plays on the raspi has short breaks in the replay. 

The logs with the SBC decoding error and the missing RTB packages still continue too.
Real-Time-Transport Protocol sounds for me like somethig that could cause this. 

in the beginning:

15.10.19 17:49:22 (+0200) <bluetooth-audio> /usr/bin/bluealsa: Unsupported RTP payload type: 1

for very many times in a row. and then...

15.10.19 18:13:13 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such file or directory
15.10.19 18:13:13 (+0200) <bluetooth-audio> /usr/bin/bluealsa: Missing RTP packet: 320 != 319
15.10.19 18:13:18 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such file or directory
15.10.19 18:13:18 (+0200) <bluetooth-audio> /usr/bin/bluealsa: Missing RTP packet: 652 != 651
15.10.19 18:13:33 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such process
15.10.19 18:13:38 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such file or directory
15.10.19 18:13:38 (+0200) <bluetooth-audio> /usr/bin/bluealsa: Missing RTP packet: 2030 != 2029
15.10.19 18:13:39 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such file or directory
15.10.19 18:13:39 (+0200) <bluetooth-audio> /usr/bin/bluealsa: Missing RTP packet: 2070 != 2069
15.10.19 18:13:52 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such process
15.10.19 18:13:54 (+0200) <bluetooth-audio> /usr/bin/bluealsa: SBC decoding error: No such file or directory

.... and so on

AlexProgrammerDE, 2019-10-15:
What happens when you run:

`apt-get update && apt-get install --reinstall alsa-utils`

In the cloud dashboard terminal in the bluetooth-audio container.

mabli, 2019-10-15:
No Change. :-(

mabli, 2019-10-15:
I found this discussion of a similar problem, which points in the direction of weak bluetooth signal.

[https://forums.balena.io/t/music-will-skip-and-or-pause-using-balena-sound/34260](https://forums.balena.io/t/music-will-skip-and-or-pause-using-balena-sound/34260)

mabli, 2019-10-15:
could this fix it perhabs?

2. Configure /etc/asound.conf (or .asoundrc in your home directory)

pcm.justboom-dac { 
type hw card 0 
}

pcm.!default { 
type plug 
slave.pcm "dmixer" 
}

pcm.dmixer { 
type dmix 
ipc_key 1024 
slave { 
pcm "justboom-dac" 
channels 2 
} 
}

ctl.dmixer { 
type hw 
card 0 
}

AlexProgrammerDE, 2019-10-16:
Try it.

mabli, 2019-10-16:
Nope, didn't change anything regarding sbc decoding error and lost packages. the pauses and stops are still there. i ordered a bluetooth dongle, to give it a try.

AlexProgrammerDE, 2019-10-16:
Ok

AlexProgrammerDE, 2019-10-16:
@mabli It seems like you have the same problem like in #23 . This seems like a bluetooth problem. VoxelTek sayed me that the RPI has no external AMP or DAC.

mabli, 2019-10-16:
Yeah, i tried also without the amp board an get the same result.

AlexProgrammerDE, 2019-10-17:
@mabli It could be a problem with your mobile phone. You have got the same hardware like everyone else and the problem appers only on your RPI and the one in #23 . It is really mysterious. I hope the bluetooth dongle will fix the problem.

mabli, 2019-10-17:
i re-installed  balena OS. This time i choose balenaOS 2.38.0+rev1 production. It appears to function now without the package loss and sbc errors. i don't know why. For me it is really a mystery. ;-)

closed 2019-10-17 by mabli

AlexProgrammerDE, 2019-10-17:
Oh great.  I will recommend your solution in #23 . Thanks.

deadlyjinx, 2020-05-03:
> i re-installed balena OS. This time i choose balenaOS 2.38.0+rev1 production. It appears to function now without the package loss and sbc errors. i don't know why. For me it is really a mystery. ;-)

Thank you! I was using version v2.47.0+rev1 (recommended) on a Raspberry Pi 3 and had the same problems. Installing version 2.38.0+rev1 fixed all the isues.