iotsound/iotsound #49: Spotify airplay & Podcasts

dimitrisnl, 2019-10-17:
I noticed that I can't stream Podcasts, although music works just fine.

I believe this could be since Podcasts are being considered "mixed" content, with both Audio & Video, although I double video is supported in the regular spotify client.

Is there any known workaround to possible stream podcasts too?

AlexProgrammerDE, 2019-10-17:
How about bluetooth? This should be a workaround for the next time.

AlexProgrammerDE, 2019-10-17:
And what do you mean with Spotify airplay & Podcast? Do you mean Spotify connect?

dimitrisnl, 2019-10-17:
@AlexProgrammerDE Yes, my bad. I meant Spotify Connect.
Bluetooth is definitely an option, just wondering.

dimitrisnl, 2019-10-17:
Closing. It's something on Spotify's side as I read through some discussions on their boards.

closed 2019-10-17 by dimitrisnl

dmnc-net, 2019-10-30:
Even this is closed issue I can confirm that you can't play podcasts thru Spotify Connect devices (at least based on librespot), only on native Spotify applications.

AlexProgrammerDE, 2019-10-30:
Do you know a alternative?

dimitrisnl, 2019-10-30:
relevant thread on librespot https://github.com/librespot-org/librespot/issues/90

AlexProgrammerDE, 2019-10-30:
I am finished. It could take the developer a long time to integrate it from librespot-java to librespot. I saw that it is not perfect implemented to people which have no premium like me but if you have premium should it work. People without premium can connect to it and in this case is the client will crash. I hope you will bring me good result when you try it. Sorry that i make it this way but my fork is at the moment in a PR. I think you can add this files by yourself.
[files.zip](https://github.com/balena-io-projects/balena-sound/files/3789885/files.zip)