iotsound/iotsound #54: Using container on other than balenaOS

ptanmay143, 2019-11-13:
Can we use this docker container to run on some other OS like Raspbian or Arch Linux ARM. I ask this because I do not want to install a new OS just to use this single docker when I have everything else configured in the other OS.

AlexProgrammerDE, 2019-11-16:
@ptanmay143 Yes it works. You can do the following steps to run the compose on boot:

Before you do the next steps you need to make sure that all neccesary thing are installed:
`curl -sSL https://get.docker.com | sh`

Then you need to run this:
`sudo usermod -aG docker pi`

Then you need to run this:
`sudo apt-get install -y libffi-dev libssl-dev python python-pip && sudo pip install docker-compose`

First you need to clone the repo:
`git clone https://github.com/balena-io-projects/balena-sound.git`

Then you need to go into the folder:
`cd balena-sound `

Then you should run pwd to know your directory path:
`pwd`
NOTE: You will need the output in the nano command

After pwd you should create a systemd service which runs the containers on boot:
`sudo nano /etc/systemd/system/BalenaSound.service`

In the following Box you need to paste the following (If the output of pwd not matches with the path of the WorkingDirectory=/home/pi/balena-sound/ you need to replace  /home/pi/balena-sound/ with the output):

[Unit]
Description=Docker Compose Application Service
Requires=docker.service
After=docker.service

[Service]
WorkingDirectory=/home/pi/balena-sound/
ExecStart=/usr/local/bin/docker-compose up
ExecStop=/usr/local/bin/docker-compose down
TimeoutStartSec=0
Restart=on-failure
StartLimitIntervalSec=60
StartLimitBurst=3

[Install]
WantedBy=multi-user.target

Then you need to save the file with shift and X at once.

You can now add the file to systemd with:
sudo systemctl enable BalenaSound && sudo systemctl start BalenaSound

capriciousduck, 2019-11-19:
It is giving  "cannot find dockerfile" error

AlexProgrammerDE, 2019-11-19:
@capriciousduck i looked at the dockerfiles and saw that some problems with building can occure on building. There is something like FROM balenalib/%%BALENA_MACHINE_NAME%%:stretch and other file names. I think you can use this file: https://send.firefox.com/download/bf7b45f71b7883e1/#oiv1-CO17bNXRQ7GTMtYYg Just drag and drop it to the bluetooth-audio folder.

capriciousduck, 2019-11-20:
The link is not working.

And can you clarify one thing...should I have to install bluetooth drivers on the host or is it not necessary?

AlexProgrammerDE, 2019-11-20:
It is not neccesary the dockerfile installs it on the container. Could you say me your model? It need to be written to the base image in the dockerfile.

AlexProgrammerDE, 2019-11-20:
https://drive.google.com/file/d/1CJnSE4T5WUGPk5nL2OPzlPoCXwrDriwu/view?usp=drivesdk

capriciousduck, 2019-11-20:
Mine is a Pi 3B.

capriciousduck, 2019-11-20:
Alex I got the repo name in FROM from docker hub.

Yesterday, I set the FROM command by setting it to balenalib of Pi3

Still  it didn't work. I'll build the image once again with a fresh install
of raspbian stretcha and will let you know.

AlexProgrammerDE, 2019-11-20:
You tried it with my Dockerfile copied and pasted in the bluetooth-audio directory of the balena-sound root directory?
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

[

{

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/balena-io-projects/balena-sound/issues/54?email_source=notifications\u0026email_token=AJXH6TEXYRBUNQ7K45L25CTQUUGVVA5CNFSM4JNBF762YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEERQDDQ#issuecomment-555942286",

"url": "https://github.com/balena-io-projects/balena-sound/issues/54?email_source=notifications\u0026email_token=AJXH6TEXYRBUNQ7K45L25CTQUUGVVA5CNFSM4JNBF762YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEERQDDQ#issuecomment-555942286",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

capriciousduck, 2019-11-20:
I built the image using the image  balenalib/raspberrypi3:stretch as the base image. And I set some environmental variable based on my preference and I got the following error:

> Error org.freedesktop.DBus.Error.ServiceUnknown: The name org.bluez was not provided by any .service files

capriciousduck, 2019-11-20:
Yeah I even tried with your docker file. I am getting the same error. And moreover I get the error that "/host/run/dbus/system_bus_socket" doesn't exist.

tmigone, 2019-11-20:
Hey @capriciousduck, when using balenaOS the label 'io.balena.features.dbus' on the docker-compose.yml bind mounts the host dbus ("/run/dbus") to the container ("/host/run/dbus"). Since you are not using balenaOS the label is doing nothing. You should try bind mounting it manually on the docker-compose file

capriciousduck, 2019-11-20:
I've rebuilt the image and deployed it. I have, as you said, mounted the dbus of my host to that of the container. When I spin up a new container I get the following error: 

```
Can't open HCI socket.: Address family not supported by protocol
Can't open HCI socket.: Address family not supported by protocol
/usr/bin/bluealsa: No HCI device available
```

tmigone, 2019-11-20:
That looks like your host OS does not have bluetooth drivers installed or enabled.

AlexProgrammerDE, 2019-11-20:
@capriciousduck you can try it with: `sudo apt update && sudo apt upgrade -y && sudo apt install -y alsa-utils bluealsa bluez python-gobject python-dbus mplayer`

capriciousduck, 2019-11-20:
Still not working.

Getting the following error:
```
Setting output volume to 85%
Connection notify volume is 85%
Restarting bluetooth service
Bluetooth agent registered
Can't open HCI socket.: Address family not supported by protocol
Can't open HCI socket.: Address family not supported by protocol
/usr/bin/bluealsa: No HCI device available
Device is discoverable as "Equinox"
/usr/bin/bluealsa-aplay: BlueALSA connection failed: No such file or directory
```

AlexProgrammerDE, 2019-11-20:
```
sudo systemctl start dbus-org.bluez.service
```
```
sudo systemctl status dbus-org.bluez
```

tmigone, 2020-01-10:
Hey @capriciousduck did you manage to get it running?

chrisys, 2020-01-27:
I'm going to close this due to lack of activity and no specific issue listed. @capriciousduck if you find specific things that need to be changed to allow this functionality by all means let us know and we can integrate them into the project.

closed 2020-01-27 by chrisys

psharma04, 2020-03-07:
> https://drive.google.com/file/d/1CJnSE4T5WUGPk5nL2OPzlPoCXwrDriwu/view?usp=drivesdk

This file link does not exist. Could someone re-upload it?

capriciousduck, 2020-04-18:
> Can we use this docker container to run on some other OS like Raspbian or Arch Linux ARM. I ask this because I do not want to install a new OS just to use this single docker when I have everything else configured in the other OS.

I used the "Dockerfile.template" dockerfile, which you can find in the bluetooth-audio, to build my own image. What I did was: I create the directories manually. 

Create /var/run/dbus/system_bus_socket manually. Not "/host/run/dbus/system_bus_socket", but "/var/run/dbus/system_bus_socket. Also change the Unix path in the dockerfile.

Mount the "/var/run/dbus/system_bus_socket" of the host to "/var/run/dbus/system_bus_socket" in the container.

That's it. That is all I did(as far as I can remember). I am not using balena-sound. I am currently using my Pi as a media center.

You try and let me know how it went.

MBMM89, 2021-05-05:
I have followed AlexProgrammerDE's instructions and need some help
am getting this from the service
● BalenaSound.service - Docker Compose Application Service
   Loaded: loaded (/etc/systemd/system/BalenaSound.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2021-05-05 19:57:58 BST; 32min ago
  Process: 1101 ExecStart=/usr/local/bin/docker-compose up (code=exited, status=1/FAILURE)
 Main PID: 1101 (code=exited, status=1/FAILURE)

May 05 19:57:57 raspberrypi systemd[1]: BalenaSound.service: Main process exited, code=exited, status=1/FAILURE
May 05 19:57:57 raspberrypi systemd[1]: BalenaSound.service: Failed with result 'exit-code'.
May 05 19:57:58 raspberrypi systemd[1]: BalenaSound.service: Service RestartSec=100ms expired, scheduling restart.
May 05 19:57:58 raspberrypi systemd[1]: BalenaSound.service: Scheduled restart job, restart counter is at 6.
May 05 19:57:58 raspberrypi systemd[1]: Stopped Docker Compose Application Service.
May 05 19:57:58 raspberrypi systemd[1]: BalenaSound.service: Start request repeated too quickly.
May 05 19:57:58 raspberrypi systemd[1]: BalenaSound.service: Failed with result 'exit-code'.
May 05 19:57:58 raspberrypi systemd[1]: Failed to start Docker Compose Application Service.
and this
pi@raspberrypi:~ $ journalctl -u BalenaSound.service
-- Logs begin at Thu 2019-02-14 10:12:00 GMT, end at Wed 2021-05-05 20:00:29 BST
May 05 00:36:43 raspberrypi systemd[1]: Started Docker Compose Application Servi
May 05 00:36:48 raspberrypi docker-compose[822]: Building audio
May 05 00:36:49 raspberrypi docker-compose[822]: unable to prepare context: unab
May 05 00:36:49 raspberrypi docker-compose[822]: Service 'audio' failed to build
May 05 00:36:49 raspberrypi systemd[1]: BalenaSound.service: Main process exited
May 05 00:36:49 raspberrypi systemd[1]: BalenaSound.service: Failed with result 
May 05 00:36:49 raspberrypi systemd[1]: BalenaSound.service: Service RestartSec=
May 05 00:36:49 raspberrypi systemd[1]: BalenaSound.service: Scheduled restart j
May 05 00:36:49 raspberrypi systemd[1]: Stopped Docker Compose Application Servi
May 05 00:36:49 raspberrypi systemd[1]: Started Docker Compose Application Servi
May 05 00:36:52 raspberrypi docker-compose[936]: Building audio
May 05 00:36:52 raspberrypi docker-compose[936]: unable to prepare context: unab
May 05 00:36:52 raspberrypi docker-compose[936]: Service 'audio' failed to build
May 05 00:36:52 raspberrypi systemd[1]: BalenaSound.service: Main process exited
May 05 00:36:52 raspberrypi systemd[1]: BalenaSound.service: Failed with result 
May 05 00:36:53 raspberrypi systemd[1]: BalenaSound.service: Service RestartSec=
May 05 00:36:53 raspberrypi systemd[1]: BalenaSound.service: Scheduled restart j
May 05 00:36:53 raspberrypi systemd[1]: Stopped Docker Compose Application Servi
May 05 00:36:53 raspberrypi systemd[1]: Started Docker Compose Application Servi
May 05 00:36:55 raspberrypi docker-compose[980]: Building audio
May 05 00:36:55 raspberrypi docker-compose[980]: unable to prepare context: unab
May 05 00:36:55 raspberrypi docker-compose[980]: Service 'audio' failed to build
lines 1-23

-- Logs begin at Thu 2019-02-14 10:12:00 GMT, end at Wed 2021-05-05 20:00:29 BST. --
May 05 00:36:43 raspberrypi systemd[1]: Started Docker Compose Application Service.
May 05 00:36:48 raspberrypi docker-compose[822]: Building audio
May 05 00:36:49 raspberrypi docker-compose[822]: unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/pi/balena-sound/core/a
May 05 00:36:49 raspberrypi docker-compose[822]: Service 'audio' failed to build : Build failed
May 05 00:36:49 raspberrypi systemd[1]: BalenaSound.service: Main process exited, code=exited, status=1/FAILURE
May 05 00:36:49 raspberrypi systemd[1]: BalenaSound.service: Failed with result 'exit-code'.
May 05 00:36:49 raspberrypi systemd[1]: BalenaSound.service: Service RestartSec=100ms expired, scheduling restart.
May 05 00:36:49 raspberrypi systemd[1]: BalenaSound.service: Scheduled restart job, restart counter is at 1.
May 05 00:36:49 raspberrypi systemd[1]: Stopped Docker Compose Application Service.
May 05 00:36:49 raspberrypi systemd[1]: Started Docker Compose Application Service.
May 05 00:36:52 raspberrypi docker-compose[936]: Building audio
May 05 00:36:52 raspberrypi docker-compose[936]: unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/pi/balena-sound/core/a
May 05 00:36:52 raspberrypi docker-compose[936]: Service 'audio' failed to build : Build failed
May 05 00:36:52 raspberrypi systemd[1]: BalenaSound.service: Main process exited, code=exited, status=1/FAILURE
May 05 00:36:52 raspberrypi systemd[1]: BalenaSound.service: Failed with result 'exit-code'.
May 05 00:36:53 raspberrypi systemd[1]: BalenaSound.service: Service RestartSec=100ms expired, scheduling restart.
May 05 00:36:53 raspberrypi systemd[1]: BalenaSound.service: Scheduled restart job, restart counter is at 2.
May 05 00:36:53 raspberrypi systemd[1]: Stopped Docker Compose Application Service.
May 05 00:36:53 raspberrypi systemd[1]: Started Docker Compose Application Service.
May 05 00:36:55 raspberrypi docker-compose[980]: Building audio
May 05 00:36:55 raspberrypi docker-compose[980]: unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/pi/balena-sound/core/a
May 05 00:36:55 raspberrypi docker-compose[980]: Service 'audio' failed to build : Build failed
lines 1-23
what did i do wrong?

lachlancresswell, 2021-10-21:
After replacing all the balena variables, renaming Dockerfiles and a few other tweaks, I managed to get everything except bluetooth up and running. The bluetooth container doesn't discover any devices (I'm using a Raspberry Pi 3 A+ with bluetooth enabled). Any hints with this last hurdle as to how to get the container to interface correctly with Raspberry Pi OS @tmigone ? Host networking is enabled and the device shows in `hciconfig` and `bluetoothctl` on the host.

tmigone, 2022-01-07:
@lachlancresswell I'm not aware of any balena specific code or requirement on the bluetooth container, so not sure what can be the problem here. Have you tried manually setting up the bluetooth connection? Here is a guide: https://pimylifeup.com/raspberry-pi-bluetooth/

ThePoonerCap, 2023-11-03:
> After replacing all the balena variables, renaming Dockerfiles and a few other tweaks, I managed to get everything except bluetooth up and running. The bluetooth container doesn't discover any [...]

Can you share the dockerfiles and  stuff? Thanks!