iotsound/iotsound #58: Add name differentiation for different services

chrisys, 2019-11-19:
Airplay and bluetooth services are named identically so in some situations it's impossible to differentiate between the two. Example from macOS:

[image]

AlexProgrammerDE, 2019-11-20:
@chrisys this can be easily fixed by defining service variables. 

If you delete the worlwide device variable and set `BLUETOOTH_DEVICE_NAME` for airplay and bluetooth-audio.

This is not a workaround for people which want to have no variable set, but it is a solution.

chrisys, 2019-11-22:
@AlexProgrammerDE good thinking! That's a good workaround

AlexProgrammerDE, 2020-01-31:
@chrisys i fixed it in #95 . I requested you to review.

AlexProgrammerDE, 2020-02-07:
@chrisys you can now close this issue. You already merged my PR.

tmigone, 2020-02-07:
Thanks again!

closed 2020-02-07 by tmigone