iotsound/iotsound #61: Newlines are in DOS format

Crinfarr, 2019-11-28:
Current version errors out with 
`$'\r': command not found`
as DOS newlines are \r\n instead of a Unix \n.  Easily fixed with find/replace.

tmigone, 2019-11-28:
Hi @Crinfarr , could you specify what file(s) are you talking about? Thank you.

Crinfarr, 2019-11-29:
Specifically the start.sh files are causing trouble, but I would suggest fixing all executed files just in case.

AlexProgrammerDE, 2019-12-01:
Hi @Crinfarr i don´t know if we have the same problem. Are you using Windows?

Crinfarr, 2019-12-03:
My computer is windows, but it shouldn't matter with a direct clone of the repo AFAIK...

AlexProgrammerDE, 2019-12-03:
Yes i did also think that until i saw that. Could you please try to dowload the Zip file (This is under the url *.git). Then you need to spawn a shell in the uzipped folder. In the shell can you push with the balena-cli like before.

VictorGamer072YT, 2019-12-10:
Yup, had to do this, else it would not work, thanks

chrisys, 2019-12-12:
@Crinfarr when you clone a repo with git, it will change the line endings to those used by your system and then switch back when you commit. See here for info: https://stackoverflow.com/questions/10418975/how-to-change-line-ending-settings

closed 2019-12-14 by chrisys