iotsound/iotsound #62: Sound cuts out or skips every 30 seconds or so, Pi 3

Nibbos, 2019-12-04:
Experiencing regular sound cutting off when using bluetooth project suggested here:
[https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/]
I added a new DEVICE CONFIGURATION variable, ‘RESIN_HOST_CONFIG_dtoverlay’ and set its value to ‘pi3-disable-wifi’ but this seems to have no effect - although I  can no longer connect to the Pi 3 from the BalenaSound interface (because the wifi on my Pi 3 has been turned off, ironically) and so cannot change anything else now. Nor do I know how to turn wifi back on again since I can't PuTTY into it either...

Can anyone suggest a possible fix to this, please? I’m semi-competent with Pi but do need simple, clear instructions to guide me.
Without a fix this glitch is just too annoying to live with.
I would buy a dedicated bluetooth receiver for my stereo but I felt that re-purposing a Pi 3 would be more appropriate and allow me to stream via a DAC slound card that I am going to add to the Pi (if I can fix the sound issue).
Thanks in advance.
Nibbos

PS please don't ask for output files from the Pi since I am no longer able to communicate with it as a consequence of turning wifi off,

AlexProgrammerDE, 2019-12-11:
I have a connection resolution for you. You can simply flash the image again. (The part with the micro sd card and BalenaEtcher) The PI will get on its own the project files back and then everything should work. But dont forget to delete the from you set var before doing everything.

stebla27, 2019-12-14:
My sound drops also on Raspi 4! In log you can see:
 bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 9953 != 9952
14.12.19 15:57:17 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 9956 != 9954
14.12.19 15:57:17 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 9964 != 9957
14.12.19 15:57:17 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 9967 != 9966
14.12.19 15:57:17 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 9973 != 9972
14.12.19 15:57:17 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 9977 != 9975
14.12.19 15:57:19 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 10068 != 10067
14.12.19 15:57:19 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 10071 != 10070
14.12.19 15:57:19 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 10105 != 10103

Not really a stable connection and unusable in this state!

AlexProgrammerDE, 2019-12-14:
This error of you @stebla27 is very similar to the one in #47 . You can try it with another balenaOS version or a bluetooth dongle.

stebla27, 2019-12-14:
> This error of you @stebla27 is very similar to the one in #47 . You can try it with another balenaOS version or a bluetooth dongle.

I use balenaOS 2.41! I can't select another version in the balena backend!
Raspi 4 has bluetooth on board! Why should I use a Bluetooth dongle?

AlexProgrammerDE, 2019-12-14:
@stebla27
1. When you download the image can you (as far as i know) select the version,at the page of the wifi  password and developement or factory option.
2. In some cases the proximity or the build in bluetooth of your mobile causes mobile. Some people solved this type of error with a bluetooth dongle.

javexed, 2019-12-16:
I can confirm this issue is with the Raspberry Pi 3+ as well. Moving to direct streaming fixed the audio quality (not really a proper solution) as that bypasses the bluetooth radio and streams over the wifi connection.

AlexProgrammerDE, 2019-12-16:
@javexed do you have logs from the dashboard?

AlexProgrammerDE, 2019-12-16:
By the way here is a link to a issue in bluealsa with the same errors: https://github.com/Arkq/bluez-alsa/issues/110

javexed, 2019-12-16:
Yes, looks similar to that issue, here are some of the logs:

08.12.19 08:05:24 (-0500)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 48438 != 48437
08.12.19 08:05:38 (-0500)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 49413 != 49410

javexed, 2019-12-16:
Other important note, balena diagnostics also show under voltage. Am looking for a better power supply to see if it is related to this issue.

AlexProgrammerDE, 2019-12-16:
@javexed Good idea!

Jordan-Adelson, 2019-12-21:
Having the same issue myself. Bluetooth connection from a Huawai Mate 10 to RPI 3. Noted that some users suspected this was an iOS issue, likely not limited to iOS. 

These are the two errors repeated in my logs.
21.12.19 22:35:35 (+1100)  bluetooth-audio  /usr/bin/bluealsa: SBC decoding error: No such process
21.12.19 22:35:35 (+1100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 27632 != 27630

Have tested moving the devices closer together and further apart, seems to have little impact on the errors until there is a considerable distance between them. 

Is there any sort of buffer that can be turned on?

Have not tried turning wifi off yet noticing that many others have not seen any improvement. 

Seems to defeat the purpose of the project a bit if having to get an additional Bluetooth dongle is the solution.

perseus177, 2020-01-05:
Sound cuts out or skips every 30 seconds
Same problem when I am using Samsung Galaxy S10+

No problem when I am using iPhone 6S

tmigone, 2020-01-20:
Hello,

Stuttering issues on the Raspberry Pi 3 are related to a known problem with the BT/WiFi chip. Both the chip manufacturer (Cypress) and the Raspberry Pi Foundation have acknowledged this but unfortunately it does not seem to be a high priority task for them (and I asume it's less likely to happen since the Pi 4 came out and does not have this problem).

The current "workaround" is to use a bluetooth usb dongle and disable the Pi's internal bluetooth, as long as the bluetooth dongle gets assigned the device name `hci0` you should be fine (which in my experience is always, but let me know if it's not). 

We realize having to buy a bluetooth adapter for a brand new Raspberry Pi is not an optimal solution so we've reached out to the foundation to see if there is anything we can do to help; however I don't anticipate a resolution to this problem anytime soon.

In case anyone is curious, here are the relevant GitHub issues:
- https://github.com/raspberrypi/linux/issues/1552
- https://github.com/raspberrypi/linux/issues/1444
- https://github.com/Arkq/bluez-alsa/issues/60

For future reference, here is the typical log output you get when experiencing this issue: 

````
20.01.20 18:14:30 (+0400)  bluetooth-audio  /usr/bin/bluealsa: SBC decoding error: No such process
20.01.20 18:14:30 (+0400)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 7047 != 7034
20.01.20 18:14:32 (+0400)  bluetooth-audio  /usr/bin/bluealsa: SBC decoding error: No such process
20.01.20 18:14:35 (+0400)  bluetooth-audio  /usr/bin/bluealsa: SBC decoding error: No such file or directory
20.01.20 18:14:36 (+0400)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 7433 != 7430
20.01.20 18:14:37 (+0400)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 7536 != 7535
```

damaev, 2020-02-18:
Hi everybody,

I really spend a lot of time trying to figure out the reason for my stuttering audio (with wifi off) while a working solution for my issue has already been released years ago: reducing the baudrate to 460800. (I was always thinking about increasing it...) See original post here: https://github.com/raspberrypi/linux/issues/2264#issuecomment-344911712

What to do:
edit `/usr/bin/btuart`  (loaded during boot by hciuart.service)
reduce the baud rate from `921600` to `460800` (in the only possible line)

Following additional information:
> The rev 1.2 Pi3B lacks the flow control signals to the Bluetooth modem (we ran out of pins), but the rev 1.3 board drops the BT PCM interface and hooks up the flow control. I think what you are seeing is occasional data loss when the FIFOs overflow, something which is hard to avoid in all circumstances. You may be able to lessen the problem by reducing the baudrate on the modem - try editing /usr/bin/btuart, replacing the 921600 with 460800 and rebooting."
> 
> It seems the 3B+ has the same problem described here for the first revisions of the 3B.

_Originally posted by @docgloucester in https://github.com/nicokaiser/rpi-audio-receiver/issues/38#issuecomment-581053695_

tmigone, 2020-02-19:
hi @damaev , thanks for your contribution! 

I just want to add that some time ago I tested the solution you propose and I had mixed results, it did make the stuttering less noticeable on one of my devices, but had no impact on another. If enough people report that it provides a noticeable benefit we definitely should consider patching it or at least mentioning it on the `README`.

nazrhom, 2020-03-04:
Hey @tmigone @chrisys, a user in the support thread linked above mentioned they experienced similar issues after changing the `BLUETOOTH_DEVICE_NAME` env var, and were able to resolve the issues when deleting that.

Daniol71, 2020-03-04:
Hi, been looking at this issue for a little while now and downgrading to balenaOS v2.38.0+rev1 fixed my bluetooth issues on the RPi3. Obviously a temporary fix but hopefully it helps someone identify what in the following updates broke bluetooth on the RPi3 (perhaps something that got added with RPi4 support). Hope it helps, original post [https://forums.balena.io/t/balenasound-blueooth-audio-on-raspberry-pi-3-sound-skipping-dropping/40894/40?u=cdl](https://forums.balena.io/t/balenasound-blueooth-audio-on-raspberry-pi-3-sound-skipping-dropping/40894/40?u=cdl)

MatthewCroughan, 2020-03-27:
I concur and have had the same experience as Daniol71. Bluetooth performance is much better after reverting to balenaOS v2.38.0+rev1. Though sometimes packets still do drop in an annoying manner, I can't get through a single tune without being irked by it. 

If a bluetooth adapter is required, and it seems like it is to get a baseline usable experience, that should be noted in the README.md of the repository. 

Here's a typical log from `dmesg` when playing back music over Bluetooth:

```
[  344.033690] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.033769] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.033853] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.033940] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034026] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034113] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034199] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034352] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034398] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034486] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034572] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034659] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034746] Bluetooth: hci0: Frame reassembly failed (-84)
[  344.034846] Bluetooth: hci0: Frame reassembly failed (-84)
[  376.771228] systemd-udevd[30]: seq 2173 '/devices/virtual/input/input1' killed
[  376.773983] systemd-udevd[30]: worker [94] terminated by signal 9 (Killed)
[  376.774023] systemd-udevd[30]: worker [94] failed while handling '/devices/virtual/input/input1'
```

MatthewCroughan, 2020-03-27:
Scratch that. It seems that it has randomly recovered, as if it has warmed up and is no longer dropping packets. Quite odd. The kernel log is not reporting any undervoltage via `dmesg`, which is the only thing I can think of that would cause it to be so strangely intermittent.

The power supply I'm using is an older model of [this.](https://www.amazon.co.uk/AUKEY-Quick-Charge-3-0-Charger-Black/dp/B01EFU8KR0/ref=sr_1_6?dchild=1&keywords=aukey+charger&qid=1585334309&sr=8-6) Perhaps something fishy is going on with regards to power delivery and all these "smart charging" features. I'll try with an official PSU at some point and report back.

MatthewCroughan, 2020-03-27:
As things got better, they got worse, as can be expected from power issues. After some time there were consistent undervoltages reported via dmesg, as well as a lot of `Frame reassembly failed (-84)`.

At the very top of the kernel log, at boot-time, there was a single undervoltage reported. This was not then reported continually until around 10-20 mins of usage. I should have saved the logs so this was evident, but did not do so.

I've swapped this PSU out for a Motorola 2A PSU, not in any way official or validated for the Pi. Next I'm going to try a Xioami 3A PSU, one that I got with my Xiaomi Mi 9 if there are issues.

MatthewCroughan, 2020-03-27:
Wow. I didn't expect to see such a difference between all these PSUs. I exposed each PSU to around 15 mins of testing using the same mobile device to stream audio via Bluetooth.

### Aukey 3A with "Smart Charging":
- Lots of frame reassembly failure
- Undervoltage reported every now and then

### Motorola 2A PSU:
- Less frame reassembly failure
- Bursts of reassembly failure every minute or so
- *No undervoltage was reported* from the kernel at any stage

### Motorola 3A PSU:
- Frame reassembly failure *only* occurred at the moment of connection from the mobile device
- No reassembly failure during usage, whatsoever
- No undervoltage reported

The 3A PSU seems to be providing me with a pretty much flawless experience.

AlexProgrammerDE, 2020-03-27:
@MatthewCroughan what Pi are you using?

MatthewCroughan, 2020-03-27:
After about 30 mins of playback, I just received a second burst of reassembly errors. Though this is not a problem, as I have lots of other equipment that has similar hickups and it is to be expected. It's performing better than my bluetooth earbuds for example

```
[  100.347229] systemd-udevd[29]: seq 1838 '/devices/virtual/input/input0' is taking a long time
[  220.355463] systemd-udevd[29]: seq 1838 '/devices/virtual/input/input0' killed
[  220.356826] systemd-udevd[29]: worker [99] terminated by signal 9 (Killed)
[  220.356859] systemd-udevd[29]: worker [99] failed while handling '/devices/virtual/input/input0'
[  253.528102] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.528875] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.528934] Bluetooth: hci0 ACL packet for unknown connection handle 608
[  253.528964] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529077] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529146] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529241] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529320] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529406] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529492] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529579] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529666] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529752] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529840] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.529925] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530013] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530100] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530185] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530271] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530359] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530446] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530532] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530617] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530704] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530791] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530878] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.530966] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531055] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531145] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531240] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531325] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531413] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531496] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531582] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531671] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531683] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531755] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531842] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.531928] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.533367] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.533450] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.533536] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.533619] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.533705] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.533792] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.533879] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.533966] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534052] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534139] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534224] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534312] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534398] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534487] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534573] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534659] Bluetooth: hci0: Frame reassembly failed (-84)
[  253.534747] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.902267] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.902328] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.902492] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.902527] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.902610] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.903028] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.903147] Bluetooth: hci0: Frame reassembly failed (-84)
[... 68 lines cut ...]
[ 1329.924146] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924234] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924319] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924405] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924490] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924579] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924663] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924676] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924749] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924835] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.924922] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.925009] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.925095] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.925182] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.925275] Bluetooth: hci0: Frame reassembly failed (-90)
[... cut ...]
[ 1329.925442] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.925529] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.925616] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.925702] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.926717] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.926761] Bluetooth: hci0: Frame reassembly failed (-84)
[ 1329.927171] Bluetooth: hci0: Frame reassembly failed (-84)
```
Interestingly, the udevd errors I've mentioned in #117 have also dramatically reduced when using the 3A PSU.

MatthewCroughan, 2020-03-27:
@AlexProgrammerDE I'm using a Pi3 Model B V1.2

chrisys, 2020-04-01:
@MatthewCroughan that's some awesome investigative work! I too had no idea that the PSU was playing that much of a part here

tmigone, 2020-04-29:
Quick update on this topic...
This [issue](https://github.com/balena-os/balena-raspberrypi/issues/476) is tracking progress on a solution to the stuttering problems.

theonlysinjin, 2021-01-12:
After struggling all day to get this working on my Pi 3 B+....
The solution was to downgrade to 'v2.38.0+rev1'

I was receiving these issues and downgrade solved it...
audio  E: [pulseaudio] backend-native.c: socket(SEQPACKET, SCO) Address family not supported by protocol
audio  E: [bluetooth] a2dp-codec-sbc.c: SBC decoding error (-2)
audio  E: [bluetooth] module-bluez5-device.c: Decoding error

pjmartorell, 2021-07-30:
> After struggling all day to get this working on my Pi 3 B+....
> The solution was to downgrade to 'v2.38.0+rev1'
>
> I was receiving these issues and downgrade solved it...
> audio E: [pulseaudio] [...]

I downgraded and it works better but I still have the similar errors like the ones you mentioned:
```
30.07.21 21:34:13 (+0200)  audio  E: [bluetooth] a2dp-codec-sbc.c: SBC decoding error (-3)
30.07.21 21:34:13 (+0200)  audio  E: [bluetooth] module-bluez5-device.c: Decoding error
```
When I disconnect and connect again to the device the issue is fixed temporarily but after a few seconds it fails again.