iotsound/iotsound #63: Support for Orange Pi

Guziobi, 2019-12-05:
I tried to run Balena sound on an orange pi plus 2. Many things seams to be working, but the board is not out putting any audio from the 3.5mm jack

chrisys, 2019-12-12:
@Guziobi do you know if the audio works OK on other apps when using balenaOS on the OrangePi? Just trying to figure out if this is a balenaSound issue or something larger.

AlexProgrammerDE, 2020-01-11:
@Guziobi can you send me your log putput?

tmigone, 2020-01-13:
@Guziobi a quick google search suggests that the default audio device for the Orange Pi board is the HDMI, not the 3.5mm jack. This depends of course on the OS you are running, but might be worth checking.

chrisys, 2020-01-27:
@Guziobi as we haven't heard back from you, I'm going to close this ticket for now, feel free to reopen it when you have taken a look at the suggestions/questions above.

closed 2020-01-27 by chrisys