iotsound/iotsound #64: Sound delay (approx 1 sec)

supermooshman, 2019-12-08:
I'm experiencing a sound delay of approximately 1 second making it useless to watch videos. Is this expected behavior?

Using Raspberry Pi 3b

chrisys, 2019-12-12:
@supermooshman take a look at this line https://github.com/balena-io-projects/balena-sound/blob/bcb076810271a06fc6402693afc65ea18a9d1a76/bluetooth-audio/start.sh#L57

You can try reducing the buffer time and see if that helps.

closed 2019-12-12 by supermooshman

Micinkator, 2020-02-24:
did you fix the delay? where do i find the config once the service / app is running inside the Raspberry?

cloutier72, 2020-04-29:
This Project is a major disappointment. I use an inexpensive Bluetooth adapter to connect a “Peloton” bike to an older Bose wave radio. The adapter “forgets” the pairing settings, leading me to have to reconnect the two. 
I thought that a raspberry pi 4 would offer some stability. Wrong... it qwas even worse. The peloton/raspberry pi handshake actually froze the peloton app. 

Same thing for YouTube. There’s a frustrating delay between the image and sound. 

So... in conclusion...  great use of a raspberry pi for music - only.