iotsound/iotsound #67: User want's a visualization of the audio paying on a screen

thgreasi, 2019-12-13:


AlexProgrammerDE, 2019-12-15:
Hi @chrisys and @thgreasi  
I thought about this and did some research. 👓 
balena-io-projects has a project named balena dash: https://github.com/balena-io-projects/balena-dash 
It uses this image for displaying the browser: marhoy/balena-wpe
The image is very minimalistic and has only the needed commands.
After some more research i found this project, also from balena: https://github.com/balena-io-projects/balena-wpe
It has the code for building the image above.
I also found a visualizer for this request: https://github.com/dpayne/cli-visualize
I tried to install it, but found a huge problem:
The image has no apt and no resources like make gcc autoconf and cmake for building anything.
Is there a way to build this image and to add apt in it?
If there is a way let me now.
By the way: I have no fast PC, so if you want me to build the image it wouldn´t work. :)

AlexProgrammerDE, 2019-12-15:
Found a better way: https://github.com/balena-io-playground/x11-window-manager

thgreasi, 2019-12-16:
Hey @AlexProgrammerDE ,
Which Dockerfile.template did you try and there is no `apt`? For example for the Bluetooth audo container of this repo, I expect apt to be available, since it's bases on debian strech. You will probably need to run an update first (`apt-get update && apt-get install gcc` for example), since the base images have an empty apt cache, so that their size doesn't increase with data that will eventually become outdated. 

Anyway, in balenalib base images we additionally provide the `install_packages` helper, which streamlines the process of installing a package & clearing all caches, so that the image size stays to the minumum.
You can see that this is already used in:
https://github.com/balena-io-projects/balena-sound/blob/45619d135ce0997a025ebf23a7d020b3ef70b358/bluetooth-audio/Dockerfile.template#L7-L13

thgreasi, 2019-12-16:
Sorry for the noise of accidentally closing & re-opening, is was because of a miss-click :disappointed:

AlexProgrammerDE, 2019-12-16:
Yeah, i know what you mean. In the image i mentioned in my long message is different from the one in the bluetooth container, but i found this project: https://github.com/balena-io-playground/x11-window-manager , which is very usefull for your idea. It installs a desktop and some other features. It is also compatible with the touchscreen. I am searching for a application, which implements audio visualisation. I found projects like cava, cli-visualizer and glava, but dont know how to implement them. :(

tmigone, 2020-01-22:
Closing this issue, see #71 for reasoning, specifically [this](https://github.com/balenalabs/balena-sound/pull/71#issuecomment-567917502) comment.

closed 2020-01-22 by tmigone