iotsound/iotsound #68: Airplay issues

stebla27, 2019-12-14:
I experience a big latency and audio drop outs on Raspi 4:
14.12.19 16:38:34 (+0100)  airplay  Very large sync error: 196682 frames, with delay: 284882, td_in_frames: 19213, rt: 8157406172, nt: 8157174215, current_delay: 33712, seqno: 22859, given timestamp: 3862206919.
14.12.19 16:38:34 (+0100)  airplay  Very large sync error: 196684 frames, with delay: 284884, td_in_frames: 19224, rt: 8157406172, nt: 8157174567, current_delay: 34055, seqno: 22860, given timestamp: 3862207271.
14.12.19 16:38:34 (+0100)  airplay  Very large sync error: 196684 frames, with delay: 284884, td_in_frames: 19289, rt: 8157406172, nt: 8157174919, current_delay: 34342, seqno: 22861, given timestamp: 3862207623.
14.12.19 16:38:34 (+0100)  airplay  Very large sync error: 196680 frames, with delay: 284880, td_in_frames: 19294, rt: 8157406172, nt: 8157175271, current_delay: 34685, seqno: 22862, given timestamp: 3862207975.
14.12.19 16:38:34 (+0100)  airplay  Shome mhistake shurely: very large number of frames to drop: 196680 -- setting it to 132300.

tmigone, 2020-03-27:
Hello @stebla27 , can you please try with the latest versio of balenaSound?

AlexProgrammerDE, 2020-04-10:
@stebla27 i will close this issue for now. I think this is more a shairport-sync problem and needs to be forwarded to their project. We are not developer of shairport-sync, so we cant help you here like there. Feel free to reopen if you have new problems with airplay.

closed 2020-04-10 by AlexProgrammerDE

jacobhq, 2021-07-03:
@stebla27 Did you get anywhere with this? I'm facing this issue now on the latest version of balenaSound.