iotsound/iotsound #69: Intermittent issues with streaming

javexed, 2019-12-16:
**Issue**: Noticed that sometimes the Bluetooth would not connect, sometimes that Spotify would not work as it wouldn't show up.
**Root cause**: Running the balena diagnostics quickly identified the source of the problem. Running out of space on the drive (/mnt/data).

When trying to figure out what is using up the data and found the following two folders:
/mnt/data/docker/aufs/diff (~800MB)
/mnt/data/docker/volumes/xxxxx_spotifycache (~300MB)

So, to fix this need to figure out why these folders are getting so large and how to keep them manageable.

AlexProgrammerDE, 2019-12-17:
@javexed i implemented the spotify cache. I thougt of this problem. When you are playing music on spotify it saves some song data and other things for faster music playing. This file can get really large in the future. I think we should implement a manually clearing of the cache.

AlexProgrammerDE, 2019-12-18:
@javexed here is a link to what are docker diffs: https://stackoverflow.com/questions/50662178/what-does-docker-aufs-diff-directory-contain-why-is-it-so-big

javexed, 2019-12-18:
@AlexProgrammerDE thank you for the info. Let me see what I can figure out about doing a docker prune periodically. Any thoughts on how to manually clear the cache?

AlexProgrammerDE, 2019-12-18:
Yes, maybe with a var like `CACHE_CLEAR`

javexed, 2019-12-19:
Ok, after digging around a bit, I think that the /mnt/data/docker/aufs/diff directly "should" be ok and is likely just normal. Pruning doesn't do anything as it all looks to be in use. The recommendations in the link recommend doing restarts and upgrades, which don't help in this case. That leaves the spotify cache directly and I see two options:
1. Periodically clean the cache, only way I figured out how to do this is to "purge data" from the device, though that is manual. 
2. Get an SD card that is bigger than 2GB (as I just had this lying around)

Thus, I think I will try and find a bigger SD card and go from there.

Just thinking out loud, perhaps the documentation should be updated to recommend a minimum size for the SD card.  (https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/)

AlexProgrammerDE, 2019-12-19:
Hey @chrisys . It would be nice if you paste the SD Card info from the README in your blog post here:  https://www.balena.io/blog/turn-your-old-speakers-or-hi-fi-into-bluetooth-receivers-using-only-a-raspberry-pi/ Thanks in advance.

AlexProgrammerDE, 2019-12-20:
@javexed chris forgot to write the info to the blog post. In the README is a Sandisk 8GB Extreme recommended.

javexed, 2019-12-20:
Perfect. Closing the issue (still recommend updating the blog post to prevent others from doing the same as me.)

closed 2019-12-20 by javexed