iotsound/iotsound #72: Wi-Fi connect

clerissonferreira, 2019-12-18:
Is it possible to implement this app in the project?
https://github.com/balena-io/wifi-connect

clerissonferreira, 2019-12-19:
My idea is to make a portable speaker, and wherever I go, I can easily connect it to wifi through this project, and use airplay or spotify.

AlexProgrammerDE, 2019-12-19:
@clerissonferreira this is already implemented in balena-dash. So it is easy to copy this container and paste it in this repo. :smiley_cat:

clerissonferreira, 2019-12-19:
You could do this update / implementation, or help me do it. I have no knowledge about container, just the basics of balena to be able to upload the application to rpi.

AlexProgrammerDE, 2019-12-19:
That is very easy. I think you could test the code for me. I will open today or tomorrow a PR. I will keep you informed.

AlexProgrammerDE, 2019-12-19:
@clerissonferreira you can now test the code from my PR. My files are in my fork here: https://github.com/AlexProgrammerDE/balena-sound/tree/wifi-connect

You need to do the steps before, but download/get my zip/url from the link above.

clerissonferreira, 2019-12-19:
UOOOOOLLLLL ... Very very very much thank you. Worked perfectly......

AlexProgrammerDE, 2019-12-19:
I think in some days gets my PR approved and everyone has this feature. 👍  Thank you for this issue.

tmigone, 2019-12-27:
Hello @clerissonferreira ,

It's very easy to use different projects within the same app on a balena device by using the git submodules functionality. This allows to avoid duplicating code and merging unnecessary code into repos. 

The approach @AlexProgrammerDE will absolutely work too, but due to `wifi-connect` being a bit out of scope of this particular project we decided to not include it by default. You can read more about our reasoning on the PR Alex opened: #74 .

Here is an excerpt from it which explains a bit the proposed solution:

>I created a quick demo project that showcases how to integrate two different balena projects (or really just any project) by using git submodules. I believe it's a better alternative in cases such as this one. Feel free to check it out, fork it and hack away with it on your own projects! [Here](https://github.com/balena-io-playground/balena-multiapp-project) is the link to the repo.

Cheers!

closed 2019-12-27 by tmigone