iotsound/iotsound #73: Spotify User and Password

clerissonferreira, 2019-12-18:
The project uses raspotify, in the raspotify settings there is the option to enter the username and password, if you are a subscriber, to make the device visible on Spotify Connect over the Internet. In this project how do I set this up?

AlexProgrammerDE, 2019-12-19:
Hello @clerissonferreira . This feature is at the moment not implemented. It is a nice idea. So i will try to implement this. You will need only in the future to set two enviroment variables in the cloud dashboard (You Username/E-Mail and your password). :smiley_cat:

clerissonferreira, 2019-12-19:
Thanks for listening. Glad you can help me.

AlexProgrammerDE, 2019-12-19:
Same with this again: https://github.com/AlexProgrammerDE/balena-sound/tree/spotify-login

Please dont wonder if the wifi-connect container disappears. It is not in this branch. You can use both of them, when my PR gets approved. 👍 

PS: I cant test if this works. I dont know how to do this, so you can try this.

AlexProgrammerDE, 2020-01-02:
@clerissonferreira could you please close this issue? I implemented this feature in a PR. You can read more about it in the README.

chrisys, 2020-01-07:
This feature works perfectly - closing for now. @clerissonferreira please reopen if you're still having trouble with it.

closed 2020-01-07 by chrisys

peteretep, 2020-01-28:
I'm having trouble with this - using my spotify username and password will not authenticate. 
My password has spaces in it - could that be the cause of the issue? I'll look into it tonight

peteretep, 2020-01-29:
Confirmed that a password with spaces in it does not work. I didn't bother fixing, just changed my password instead.

tmigone, 2020-01-30:
Hey thanks for letting us know @peteretep. I'm opening a separate issue to track this and get it fixed soon.