iotsound/iotsound #79: hdmi-output: not working in RPi4

tmigone, 2020-01-15:
HDMI audio is not working on Pi 4.

zchbndcc9, 2020-02-08:
Bump. I am experiencing the same issue

zchbndcc9, 2020-02-08:
Fixed by adding `RESIN_HOST_CONFIG_hdmi_mode` set to `2` in the configuration settings.

AlexProgrammerDE, 2020-04-01:
@tmigone works the comment from zchbndcc9 for you?

AlexProgrammerDE, 2020-05-02:
All fixes are included in the troubleshooting: https://landr-balenalabs-repo-balena-sound.netlify.app/docs/troubleshooting/ Closing.

AlexdrFontes, 2021-01-10:
it did not work for me with a clean install

chrisys, 2021-01-11:
@AlexdrFontes could you give any more information? I've just done a fresh deploy of the latest version of the software onto a Pi 4 here and it worked fine without any changes.

AlexdrFontes, 2021-01-11:
Hi Chris. 

I’ve manage to use balenasound and reproduce music on several speakers, but the jack(3.5mm) signal is really really weak and to play on my amps i have to crank the volume all the way up.

I bought a regular HDMI2AV transmitter, self powered and tried with raspbian to play sound on the transmitter. Everything went ok, the sound volume is much better that with the jack.

After this I flashed balenaSound and tried to connect the transmitter to the amp, but unfortunately with no luck.
I spent several hours editing config variables that other people on this forum say it would solve the issue but I had no success.

I booted the pi with the cable on HDMI 0 slot but unfortunately the sound comes ALWAYS from the jack.

Have you been able to force HDMI to pass the sound onto one of these receivers?
My Setup would be just BalenaSound flashed directly onto the sd card and connected to the amp with HDMI - > HDMI2AV transmitter - > LEFT and RIGHT sound cable to AMP.

chrisys, 2021-01-11:
@AlexdrFontes ah, I'm replying to you on the forum here https://forums.balena.io/t/hdmi-audio-output-only-on-balenasound/232278

chrisys, 2021-01-11:
This seems to be reoccuring: https://github.com/balena-os/balena-raspberrypi/issues/586

tmigone, 2021-02-12:
Hi @AlexdrFontes @chrisys, this is consequence of changes made by the Pi foundation on the linux kernel. It's still working but they changed the behaviour of the `snd_bcm2835 ` audio hardware. Let me elaborate.

Historically the `snd_bcm2835` hardware was initialised in a way that the OS would see one sound card with multiple devices (3.5mm jack, HDMI). This meant you could output audio to both audio jack and HDMI by default with no configuration required. 

As of kernel version 5.4, the Pi foundation changed this so the OS now lists two sound cards, one for the audio jack and another one for HDMI (HDMI only showing up if you got a cable plugged in on boot). The default is the audio jack, so you won't get HDMI audio unless you select it as the output interface of your choice. I believe this was made so users get to select which one they want with a nice UI dropdown on Raspberry Pi OS (see below).

Some references:
- Pi Foundation's [changes]((https://github.com/raspberrypi/linux/commit/8e6c2ab786e7b183ca0b652ce67ff061eae20fce)) to the kernel
- balenaOS 2.54.3+rev1 incorporating those [changes](https://github.com/balena-os/balena-raspberrypi/commit/134f7d727be1275de64ced4417c373eff08611e4) **only** for the Pi4. This explains why it "doesn't work" on 2.60 but it does in 2.50.
- Pi official [docs](https://www.raspberrypi.org/documentation/configuration/audio-config.md) on how to change output audio device
- Related GitHub issues in the audio block repo with further details: [#39](https://github.com/balenablocks/audio/issues/39) and [#4](https://github.com/balenablocks/audio/issues/4)

So, how do we select which audio interface we want to use now? 
Easy, use the `AUDIO_OUTPUT` environment variable and set it to `RPI_HDMI0` for example for HDMI0 on the Pi4. This is described in detail in the balenaSound [docs](https://sound.balenalabs.io/docs/customization#general) and the audio block [readme](https://github.com/balenablocks/audio#environment-variables).

I hope this helps @AlexdrFontes, I'll reply to you in the forums too with this information so that others can find it there too. Let's continue the conversation over there if needed. Thanks!

closed 2021-02-12 by tmigone