iotsound/iotsound #80: Balena bluetooth-audio can't be discovered by Nvidia Shield TV and LG TV

qkle, 2020-01-17:
BalenaSound on my Raspberry Pi 3B+ works perfectly fine when connect via Bluetooth from any smart phones or PCs. However, it can't be discovered on the Nvidia Shield TV (2017 model) as well as my LG smart tv.
 
The Shield TV and the LG TV have no problem connecting to other Bluetooth devices (e.g. bluetooth speakers, headphones or game controller). I have tried reseting all settings and unpairing/pairing but still can not search for balenaSound device.

Is there anyone know what may cause the problem and how to fix that?

Thank you in advance!

tmigone, 2020-01-18:
Hi @qkle are you using balenaSound in Legacy Mode (with BLUETOOTH_PIN_CODE)?

qkle, 2020-01-19:
Hi @tmigone, I'm not sure how to check it but I guess it's running in SSPM mode since I do not have to enter PIN code when pairing from my phone and laptop.

tmigone, 2020-01-22:
Can you try setting it to legacy mode? You can do that by defining a pin code, see [here](https://github.com/balenalabs/balena-sound#set-bluetooth-pin-code-optional) for more details. It might be that the devices don't support sspmode, though they are not that old.

qkle, 2020-01-31:
Thanks for your suggestion but I don't think it's the case as both my TV and Nvidia Shield were able to connect to all other bluetooth speakers/headphones without problems.

Anyway I'll close this issue as I gave up and purchased a cheap bluetooth receiver and it works fine with both devices.

closed 2020-01-31 by qkle

jtonfat, 2020-03-03:
Hi! I also have the same issue with an LG projector. I have also a bluetooth dongle but I don't know how to configure the RPi to select this dongle as the bluetooth device. Could you please explain me? Thanks