iotsound/iotsound #81: Request: Add native deezer support

the4anoni, 2020-01-20:
Like in topic.

chrisys, 2020-01-27:
@the4anoni do you know if Deezer offer a Linux client that would make this possible? My very brief Goolging suggests they don't, in which case we wouldn't be able to add this.

the4anoni, 2020-01-28:
There isn't native deezer player, but there is deezer web version, or there is SMLoadr.

AlexProgrammerDE, 2020-04-09:
@the4anoni i like the idea too, but since dezzer support not a comfortable way of playing music from your mobile phone it can be only used via the web. BalenaSound points not in that direction. But i have a good idea for you. Balena has made a web browser for the Pi: https://github.com/balenalabs/balena-wpe 
You can use it to connect to `deezer.com`. And if you like to use balenaSound and this browser have a look at this guide: https://www.balena.io/blog/two-projects-one-device-turn-your-raspberry-pi-into-a-multitool/

I am closing for now. If you have questions can you comment here.

closed 2020-04-09 by AlexProgrammerDE