iotsound/iotsound #82: A user reported that balena-sound had issues working on our 64-bit beta OS

anujdeshpande, 2020-01-21:
See - https://jel.ly.fish/fbd336b3-c2c5-4e28-aefb-2c1f9daec5fe - for more info

chrisys, 2020-01-27:
@anujdeshpande could you add the information in the Github issue - Jellyfish isn't public so contributors won't be able to see what the problem is.

dtischler, 2020-02-03:
While attempting this on a Raspberry Pi 3B+ with 64-bit OS, this appears to repeat indefinitely in the logs:  

`03.02.20 14:42:59 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:00 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:00 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:01 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:01 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:02 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:02 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:03 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:03 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:04 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:04 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:05 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:05 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:06 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:06 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:07 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:07 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:08 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:08 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:09 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:09 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:10 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:10 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
03.02.20 14:43:11 (-0700)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
03.02.20 14:43:11 (-0700)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device`

gakonst, 2020-03-01:
Just a +1 on the Bluetooth issue above being reproduced on RPI 3B 64bit

chrisys, 2020-04-01:
As a fix whilst we look into this, you should be able to simply remove the `vc4-kms-v3d` overlay from the device configuration and reboot, after that the 64-bit OS should work.

andyzasl, 2020-04-12:
> As a fix whilst we look into this, you should be able to simply remove the `vc4-kms-v3d` overlay from the device configuration and reboot, after that the 64-bit OS should work.

This helped for me, thanks

theonlysinjin, 2021-01-12:
Downgrading to v2.38.0+rev1 solved this for me.
https://github.com/balenalabs/balena-sound/issues/62#issuecomment-758763978

FrozenAtlas, 2022-05-23:
This issue is still present with me on a Raspberry Pi 4 64bit on balenasound v3.9.1. Works just fine on my Raspberry Pi 3 32bit.

The workaround reference in this issue fixed the no audio problem on my Raspberry Pi 4.

> As a fix whilst we look into this, you should be able to simply remove the `vc4-kms-v3d` overlay from the device configuration and reboot, after that the 64-bit OS should work.

pjmartorell, 2023-02-10:
> As a fix whilst we look into this, you should be able to simply remove the `vc4-kms-v3d` overlay from the device configuration and reboot, after that the 64-bit OS should work.

I can confirm that removing it from DT overlays in a Raspberry Pi 3 (using 64bit OS) fixed the issue:
[image]

maggie0002, 2023-02-15:
@pjmartorell would you be able to add a PR to update the docs?