iotsound/iotsound #84: Add read-only option

slambert, 2020-01-26:
I've added this to RPis I use for streaming to make them more indestructible.

https://learn.adafruit.com/read-only-raspberry-pi/overview

May not need to be included in the software as default, but mentioned in the instructions with a link as an option?

tmigone, 2020-01-27:
Hi @slambert, thank you for your suggestion. 
balenaOS already uses a read only fs by default, all of the partitions that can be read-only are already in that mode. You can read more about it [here](https://www.balena.io/os/docs/architecture/#Stateless-and-Read-Only-rootFS).

chrisys, 2020-01-27:
Closing as we already have this handled by using balenaOS, but thanks for the suggestion @slambert 👍

closed 2020-01-27 by chrisys

slambert, 2020-01-28:
Huh... didn’t realize. Can you say more? Are you saying it solves the problem because balena-os enables remote shutdown, or because it’s a read-only? With balena-os can you unplug the system while it’s running (without a shutdown) and not risk damaging the file system?

Steve

tmigone, 2020-01-28:
Hi @slambert, yes you are correct. balenaOS uses a read-only file system so you don't have risk of damaging it.

slambert, 2020-01-29:
That's great. Thanks for your thoroughness!