iotsound/iotsound #85: Suptronics X400 DAC

peteretep, 2020-01-26:
My system is a raspberry pi 3 with a Suptronics X400 DAC - http://www.raspberrypiwiki.com/index.php/X400_V3.0
I have disabled the on-board audio by editing the existing `RESIN_HOST_CONFIG_dtparam` variable to set ”audio=off”.
I added the additional custom configuration variable called `BALENA_HOST_CONFIG_dtoverlay` with a value of `hifiberry-dac`. 
I suspect this is not the correct value.

On the volumio project, there is a list of DAC's that are supported. https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/i2s_dacs/dacs.json
The suptronics X400 has the following settings:
```    
{"id":"st400-dac-amp","name":"ST400 Dac (PCM5122) - Amp","overlay":"iqaudio-dacplus","alsanum":"1","mixer":"Digital","modules":"","script":"","needsreboot":"yes"},
```

Can anyone advise?

peteretep, 2020-01-27:
This forum thread may be a more appropriate place for this discussion. Please feel free to close this issue if so.
https://forums.balena.io/t/regarding-dac-installation-on-balenasound-project/45568/5

tmigone, 2020-01-27:
Hi @peteretep, yes, the forum is a more appropriate place in this case. If it comes down to balenaSound needing a new feature/fix then support will open an issue in here (or feel free to do so). Closing this for the time being!

closed 2020-01-27 by tmigone