iotsound/iotsound #86: Using Balena-Sound without an Internet connection

bstan2018, 2020-01-27:
On my setup, Balena-Sound appear to be ready for connection if it is connected to the Internet. Without Internet (for my case, network cable not connected), the device is not discoverable.
I understand that the initial push from cloud for setup must have Internet. However, subsequence daily usage streaming via Bluetooth should not require any Internet connection. 
As my Hifi is not co-located with good Internet, would appreciate if someone may be able to advise if there is any configuration that allow my pi/Balena-Sound to work without Internet.

Thanks!

tmigone, 2020-01-27:
Hi @bstan2018, bluetooth audio should work without internet. Are you hitting any errors or is it just that you don't see the device listed?

bstan2018, 2020-01-27:
Thank you.
I’m actually using an iphone. So, it might be AirPlay instead of Bluetooth audio. Is Internet required for airplay?
Anyway, I’ll verify again later today and update here.

Send from Outlook<https://aka.ms/qtex0l> for iOS

tmigone, 2020-01-28:
AirPlay uses network/LAN connectivity by default, so you need to have both of your devices connected to the same network. In your case it would be your phone (which is connected via WiFi), and your balenaSound device (connected either by ethernet cable or WiFi).

So I think this could explain why you can't discover the device when you don't have the network cable connected.  

One thing I want to clarify though is that balenaSound *does not* require internet access, just access to your local network. Your iPhone would be doing the streaming.

One last comment I have is that I think you can do AirPlay over bluetooth, but that is something that you would need to configure on your iPhone and unfortunately I don't have one to test it.

tmigone, 2020-02-12:
Closing this as it appears there is no issue. Feel free to reopen.

closed 2020-02-12 by tmigone

marcgarciamarti, 2020-04-05:
Hello! can we reopen this issue? I too have an iphone and I just disconnected the ethernet cable. Balena is correctly displayed as bluetooth device and I can connect to it. When I do that I would hear a ding but whatever I do, no sound comes out of my Raspberry Pi. If I hook the Pi up to the internet it will resume playing through Spotify Connect just fine.

AlexProgrammerDE, 2020-04-05:
@marcgarciamarti do i understand that correctly? Your Pi gives no output over bluetooth if it isnt connected to the network and it gives output over bluetooth if it is connected to the network?

marcgarciamarti, 2020-04-05:
Allow me some time. I’ve just tampered with this: https://forums.balena.io/t/balena-sound-low-output/61927?page=2 and while testing, I would say that I may be wrong. I’ll get back.

marcgarciamarti, 2020-04-11:
> Allow me some time. I’ve just tampered with this: https://forums.balena.io/t/balena-sound-low-output/61927?page=2 and while testing, I would say that I may be wrong. I’ll get back.

Hi again, it seems to work. I thought bluetooth didn't work when it actually had to do with the volume defined. I've set it to 100% and disabled multi-room as per my understanding of the post above and it now seems to work well. Many thanks!