iotsound/iotsound #87: Unable to use Spotify while BT is paired/connected

richbayliss, 2020-01-27:
From https://forums.balena.io/t/59446

I’ve just configured my RPi3 B+ to cast from various devices to my amplifier.
I am using bluetooth-audio to play music from the Google Home
and using Spotify Connect to play music from any other device.

When I play music from Google Home it is not possible to switch to Spotify Connect.
I get the following error messages.

```
    26.01.20 12:35:38 (+0100) spotify ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function ‘snd_pcm_open’ failed with error ‘EBUSY: Device or resource busy’
    26.01.20 12:35:38 (+0100) spotify ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
```

I can reconnect via Spotify Connect when I stop the bluetooth-audio service.

Is there a way that I can switch between services without enabling/disabling?

AlexProgrammerDE, 2020-04-10:
@richbayliss i think multiroom could solve this issue.

tmigone, 2020-09-21:
As of v3.0, balenaSound now supports simultaneous audio streams and does audio routing in a much more robust way. Should be possible to switch seamlessly between Blutooth/Spotify and even have both play at the same time.

closed 2020-09-21 by tmigone