iotsound/iotsound #90: spotify: Authentication fails if pasword has spaces in it

tmigone, 2020-01-30:
Confirmed that a password with spaces in it does not work.

AlexProgrammerDE, 2020-01-30:
Found the issue:

https://github.com/balenalabs/balena-sound/blob/abb17686d50aac16c1caf1b7909bfcf706a5c423/spotify/start.sh#L11-L14

https://github.com/balenalabs/balena-sound/blob/abb17686d50aac16c1caf1b7909bfcf706a5c423/spotify/start.sh#L17

The password is not saved in: "" which is then seen as a separate command line argument.

AlexProgrammerDE, 2020-01-30:
@tmigone i opened now a PR. I am not able to test it. Could you do it please?

AlexProgrammerDE, 2020-02-07:
@tmigone my PR #91 got merged. You can now close this issue.

closed 2020-02-07 by tmigone

Baryczka, 2020-04-22:
This PR could caused other issue:
https://github.com/balenalabs/balena-sound/issues/131