iotsound/iotsound #94: Audio distortion when volume is 100%

gthb96, 2020-01-31:
Hello, when I leave the volume level at 100% I get distorted audio on some tracks.
When I lower the volume on the phone or set the SYSTEM_OUTPUT_VOLUME to 98 the distortion goes away.

Is there any one else with this issue or a fix for it other than setting the volume level lower?

stew0ralph, 2020-02-10:
Not an answer to your question - but what command do I need to use to adjust the SYSTEM_VOLUME_OUTPUT? Mine (Raspi 3B+) seems to be extremely quiet when using Bluetooth, and quieter than I would prefer when using Spotify. 
Thanks in advance.

cenix102, 2020-03-05:
@stew0ralph I had the same problem. If you are you using the balenaCloud just go to your device and add a "Device Variable". 
name: SYSTEM_OUTPUT_VOLUME 
value: 99

that all! enjoy your music
[image]

tmigone, 2020-03-27:
I think hitting some distortion with volumes maxed out is to be expected (to some degree). As was suggested, `SYSTEM_OUTPUT_VOLUME` variable can be used to adjust overall volume. We are looking into better volume control but for the moment this is the alternative. I'll close this issue for now, feel free to reopen if you have anything else to add.

closed 2020-03-27 by tmigone