iotsound/iotsound #96: No audio output using onboard 3.5mm audio jack over bluetooth

VoxelTek, 2020-02-01:
Hi there,
So I just did this project on a RPi3, and found out that I could not play any music over Bluetooth or Spotify connect. I am able to select the device as a source in Spotify, but no music comes out of the speakers. If I connect my huawei nova 3i up to it over Bluetooth, no audio comes out of the speakers either. Looking at the console on my balenaDashboard, I see this when I try and play audio through Bluetooth:

`01.02.20 18:31:55 (+1000)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
01.02.20 18:32:37 (+1000)  bluetooth-audio  ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
01.02.20 18:32:37 (+1000)  bluetooth-audio  /usr/bin/bluealsa-aplay: Couldn't open PCM: No such device
01.02.20 18:32:42 (+1000)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 256 != 3
01.02.20 18:32:43 (+1000)  bluetooth-audio  /usr/bin/bluealsa: SBC decoding error: No such file or directory`

However, when I play from Spotify, I get this:

`01.02.20 18:36:06 (+1000)  spotify  INFO:librespot_playback::player: Loading track "Throw Your Arms Around Me" with Spotify URI "spotify:track:3T0ggmh35mdaTCXT2c0wkb"
01.02.20 18:36:47 (+1000)  spotify  INFO:librespot_playback::player: Track "Throw Your Arms Around Me" loaded
01.02.20 18:36:47 (+1000)  spotify  ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
01.02.20 18:36:47 (+1000)  spotify  ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENODEV: No such device'
01.02.20 18:36:47 (+1000)  spotify  ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed
01.02.20 18:36:47 (+1000)  spotify  ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
01.02.20 18:36:47 (+1000)  spotify  ERROR:librespot_playback::audio_backend::alsa: Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'ENODEV: No such device'
01.02.20 18:36:47 (+1000)  spotify  ERROR:librespot_playback::player: Could not start audio: Alsa error: PCM open failed`

I do not have any other RPis to test this on, but I know this:
1. The speakers that I am playing through work, and so does the cable.
2. My Spotify account is a premium account, and plays the song on other devices, including a google home.
3. The Pi's GPIO may be broken, because I cannot output anything to any of them. I can only get the 3.3v and 5v out. However, when I ran a program that tested if any of the GPIO pins were broken, it said that all of them were working.
4. A while ago, I tried a balena project that did a similar thing on the same Pi, but only with bluetooth, and it didn't work either, and gave similar messages.
5. I have no other RPis, I don't know anyone I can borrow one from, and I cannot buy one for reasons that you do not need to know.

Thanks,
VoxelTek

tmigone, 2020-02-03:
hi @VoxelTek, what version of balenaOS are you using? Are you running the 32 bits OS?

Chaosflori25, 2020-02-08:
I tried the same things, but none of them worked, I'm using the rbp3 b+  with wifi with the 64-bit version and the newest update of balena cloud and sound.

VoxelTek, 2020-02-08:
> hi @VoxelTek, what version of balenaOS are you using? Are you running the 32 bits OS?

Sorry for the late reply, but I am running the 64-bit version, although I have used the 32-bit version in the past on the same Pi, with the same errors.

tmigone, 2020-02-12:
hey @VoxelTek , I think this issue is related to the 64-bit version. I don't think i've seen it on 32-bit. I'd advise you to try with the 32-bit version again.

VoxelTek, 2020-02-13:
> hey @VoxelTek , I think this issue is related to the 64-bit version. I don't think i've seen it on 32-bit. I'd advise you to try with the 32-bit version again.

Ok, thanks for letting me know. I will try it, and if it doesn't work, I'll let you know.

VoxelTek, 2020-02-13:
Hi there,
I cannot seem to find the 32-bit version anymore, as adding a new device presents me with 2 RPi Options:
Raspberry Pi 3 (using 64-bit OS)
Raspberry Pi 4

Has something changed?

lucasdcrk, 2020-03-20:
I have the same issue running balenaOS balenaOS 2.47.1+rev1 on a Raspberry Pi. 
No audio output whatsoever, cannot find how to download the 32bit version

sebastianperruolo, 2020-03-21:
Hi, when you create an Application it will ask you what version you want, if you choose the 64bit version then it will only let you download a 64bit version for your device. I faced the same issue and I had to go back and create a new Application.

lucasdcrk, 2020-03-21:
> 
> 
> Hi, when you create an Application it will ask you what version you want, if you choose the 64bit version then it will only let you download a 64bit version for your device. I faced the same issue and I had to go back and create a new Application.

I tried this but there is no 32-bit version for the Raspberry pi 4, I tried using the 32-bit raspberry pi 3 version but it doesn't boot

tmigone, 2020-03-27:
Hi @lucasdcrk you are correct there is no 32bit version for the Pi 4. Can you share logs that show the error you are having?

VoxelTek, 2020-04-15:
Hi there,
I am unfortunately unable to test out the 32bit version as my Pi is currently unavailable for use. I am working toward getting another, but until then, I am unable to help with this problem.

VoxelTek, 2020-04-15:
Also, I believe that the error may have been related to that Pi specifically as strangely, the gpio pins had not been working, no matter what I tried. I ran a test program I found, and it said that none of my pins were burnt out, so that's strange.

meechos, 2020-04-18:
@VoxelTek same issues unfortunately. Tried both
- Raspberry Pi 3 (using 64bit OS)
- Raspberry Pi 3 
 with `balenaOS 2.47.0+rev1`.  Error log below. @tmigone any suggestions on this would be great!

```bash
18.04.20 21:35:37 (+0100)  snapcast-client  2020-04-18 20-35-37 [Notice] (start) Connected to 192.168.178.67
18.04.20 21:35:37 (+0100)  snapcast-client  2020-04-18 20-35-37 [Info] (getMacAddress) My MAC: "b8:27:eb:76:1a:55", socket: 6
18.04.20 21:35:37 (+0100)  snapcast-client  2020-04-18 20-35-37 [Info] (onMessageReceived) ServerSettings - buffer: 1000, latency: 0, volume: 100, muted: 0
18.04.20 21:35:37 (+0100)  snapcast-client  metadata:{"STREAM":"default"}
18.04.20 21:35:37 (+0100)  snapcast-client  2020-04-18 20-35-37 [Info] (onMessageReceived) Codec: flac
18.04.20 21:35:37 (+0100)  snapcast-client  2020-04-18 20-35-37 [Notice] (state) sampleformat: 44100:16:2
18.04.20 21:35:37 (+0100)  snapcast-client  ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
18.04.20 21:35:37 (+0100)  snapcast-client  2020-04-18 20-35-37 [Error] (onException) Controller::onException
18.04.20 21:35:39 (+0100)  snapcast-client  2020-04-18 20-35-39 [Warn] (sendRequest) timeout while waiting for response to: 399, timeout 2000
18.04.20 21:35:39 (+0100)  snapcast-client  2020-04-18 20-35-39 [Error] (worker) Exception in Controller::worker(): Can't open default:CARD=vc4hdmi PCM device: No such device
18.04.20 21:35:39 (+0100)  snapcast-server  2020-04-18 20-35-39 [Error] (StreamSession) Error reading message header of length 0: End of file
18.04.20 21:35:39 (+0100)  snapcast-server  2020-04-18 20-35-39 [Info] (onDisconnect) onDisconnect: b8:27:eb:76:1a:55
18.04.20 21:35:39 (+0100)  snapcast-server  2020-04-18 20-35-39 [Error] (StreamSession) Error in socket shutdown: Transport endpoint is not connected
18.04.20 21:35:40 (+0100)  snapcast-client  2020-04-18 20-35-40 [Notice] (start) Connected to 192.168.178.67
18.04.20 21:35:40 (+0100)  snapcast-client  2020-04-18 20-35-40 [Info] (getMacAddress) My MAC: "b8:27:eb:76:1a:55", socket: 6
18.04.20 21:35:40 (+0100)  snapcast-client  2020-04-18 20-35-40 [Info] (onMessageReceived) ServerSettings - buffer: 1000, latency: 0, volume: 100, muted: 0
18.04.20 21:35:40 (+0100)  snapcast-client  metadata:{"STREAM":"default"}
18.04.20 21:35:40 (+0100)  snapcast-client  2020-04-18 20-35-40 [Info] (onMessageReceived) Codec: flac
18.04.20 21:35:40 (+0100)  snapcast-client  2020-04-18 20-35-40 [Notice] (state) sampleformat: 44100:16:2
18.04.20 21:35:40 (+0100)  snapcast-client  ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
18.04.20 21:35:40 (+0100)  snapcast-client  2020-04-18 20-35-40 [Error] (onException) Controller::onException
18.04.20 21:35:40 (+0100)  snapcast-server  2020-04-18 20-35-40 [Notice] (handleAccept) StreamServer::NewConnection: 192.168.178.67
```

dextor, 2020-06-01:
> Hi @lucasdcrk you are correct there is no 32bit version for the Pi 4. Can you share logs that show the error you are having?
@tmigone switching to the 32bit version fixed this issue for me on a pi 3B

dextor, 2020-06-01:
> hey @VoxelTek , I think this issue is related to the 64-bit version. I don't think i've seen it on 32-bit. I'd advise you to try with the 32-bit version again.

@tmigone switching to the 32bit version fixed this issue for me

tmigone, 2020-07-17:
Should be addressed by #82 , feel free to reopen if you think otherwise!

closed 2020-07-17 by tmigone