iotsound/iotsound #100: Add GPIO output

laurensoosting, 2020-02-06:
Add GPIO output which can be used to enable an amplifier when music is playing.

tmigone, 2020-02-07:
Hi @laurensoosting this feature is already supported via `bluetooth scripts`:
- https://github.com/balenalabs/balena-sound#bluetooth-scripts-optional

Is this useful to you?

laurensoosting, 2020-02-07:
Im mainly looking for this feature to use with spotify. I hink this could be a nice adition. On this moment the amplifier is always on.

tmigone, 2020-02-07:
I see, in that case, yes, that is currently not possible. But it's a nice addition so we'll add it to the backlog. Thanks for the suggestion!

AlexProgrammerDE, 2020-02-07:
@tmigone and @laurensoosting i found this wiki page of librespot: https://github.com/librespot-org/librespot/wiki/Options . It has an option named onevent. I will do some research now.

tmigone, 2020-07-17:
This will be possible when 3.x is released with play/stop event hooks.

AlexProgrammerDE, 2020-12-31:
This seems fixed then?

closed 2020-12-31 by AlexProgrammerDE

shawaj, 2021-01-06:
@AlexProgrammerDE reckon it is worth adding some documentation about how to do this? I am not entirely sure of how to use the hooks. Haven't played around with it.

AlexProgrammerDE, 2021-01-06:
@shawaj yeah docs on this would be good. @tmigone can you give us a look into how this works now?

tmigone, 2021-02-09:
Added to the TODO in #393