iotsound/iotsound #102: possible to add Google Cast-ability ?

bodecove, 2020-02-08:
is it ..possible to add Google Cast-ability ?.. if so how.. I'm kind of a noob

AlexProgrammerDE, 2020-02-08:
@bodecove this feature is possible, but requires registration in the dashboard of google.
I found some sdk and guides directly from google: https://developers.google.com/cast/
I think this is a thing for another project. 

Hope it helps 
Alex

AlexProgrammerDE, 2020-02-09:
@bodecove i have bad news. :( If you want to register your account in the dashboard it wants to pay you a registration fee of 5$. Here is it: https://cast.google.com/publish/#/signup
I think that is a huge problem.

bodecove, 2020-02-12:
@AlexProgrammerDE ...i agree..  but if it can be done...I will test it ..for the 5bux...i just need to figure out how  lol

AlexProgrammerDE, 2020-02-13:
@bodecove i found this guide: https://codelabs.developers.google.com/codelabs/cast-receiver/ 
I think if this works it is a step into the right direction. 👍

bodecove, 2020-02-17:
Sorry.. Been busy.. I'll get it done this week., forward me the test.apk
again if ucan

AlexProgrammerDE, 2020-03-27:
@bodecove are you still interested in doing this? You have not replied in a while.

bodecove, 2020-04-03:
I am still interested ..just with the virus here in usa Illinois..been
really busy with news.. And many life changes..i will try and get going on
this...reply with . APK u want me to test

AlexProgrammerDE, 2020-04-04:
@bodecove i think you don´t need a apk file. It should also work with Youtube and other streaming enabled services. Can you try to get this running on a Pi: https://codelabs.developers.google.com/codelabs/cast-receiver/#0 ?

This should give you video and music output. Let me know if you are finished.

AlexProgrammerDE, 2021-03-29:
I just bought the chromecast license myself and gonna experiment with it a bit. But even if it would work I only see this inside another project. This basically requires the need for balena sound at all, but is also paid. I will give an update on this once I get this working in audio only.

jayjr1105, 2021-06-09:
Any progress on this?  It would be a dealbreaking feature for me.  I'd trash several echo dots running to various sound equipment if this picked up some traction.

JaragonCR, 2021-06-11:
I was going to ask the same, with Spotify prices on the ramp, YT Music has real traction plus amazon music as well, but we have to rely in good ole BT.

pjmartorell, 2021-11-02:
Any update on this @bodecove or @AlexProgrammerDE? I think it would be a very interesting feature even if we need to pay a few bucks

bodecove, 2021-11-05:
@pjmartorell    life has put this on back burner..i have not pursued this for some time..but yes it would be a nice feature

tommyzat, 2021-12-28:
I'd also be willing to pay for the chromecast licensing fee if necessary in order to have this. Youtube Music is getting pretty big.

Syco54645, 2022-02-06:
This could bypass the bluetooth issues on the rpi3 and also give a more stable way to stream audio than bluetooth. Bluetooth in my house has a range of about 15 feet.

Syco54645, 2022-03-22:
> I've started to look into this a little - ALthought dislaimer I'm a complete n00b. 
> Ran through this  tutorial https://developers.google.com/cast/codelabs/cast-receiver#0 as suggested by @AlexProgrammerDE ...however I think If I'm not mistaken - this requires you to have a cast device to configure against...not sure this is possible as the Pi is not a Chrome-cast but maybe I'm wrong here....
> 
> A possible creative solution is just simply plugging a chrome-cast into the HDMI slot...but that kinda defeats the purpose I would think.
> 
> Looking further it might be possible to make a web receiver and have  BalenaOS run a local web-server for 
> https://developers.google.com/cast/docs/web_receiver/basic
> More info here: 
> https://developers.google.com/cast/docs/web_receiver/core_features#creating-a-customized-receiver-app
> 
> I'm honestly probably not the guy to do this but I'm genuinely interested in learning about this.
> 

I am on mobile so my response will be short, I am not meaning to come off as rude. With that out of the way the HDMI is output. I don't think the pi support HDMI in. My solution was to use a hifiberry that has line in and use a chromecast audio. Unfortunately it seems the line in is broken on balenaSound. I ended up just using snapcast directly on an rpi running debian and it works great. 

Further I believe Google changed the cast API and afaik it has not been reversed so getting something implemented may be hard. Afaik there are no more open source Chromecast clients that are working.

bodecove, 2022-05-03:
Thanks for digging into this issue!!

tommyzat, 2022-09-04:
The chromecast "receiver" guide by Google seems to be actually a chromecast remote guide. Correct me if I'm wrong. 

https://play.google.com/store/apps/details?id=com.softmedia.receiver.castapp

This is the only app I've found to have true chromecast receiver functionality, it allows any device to cast to the phone.

tommyzat, 2022-09-04:
The chromecast "receiver" guide by Google seems to be actually a chromecast remote guide. Correct me if I'm wrong. 

**This app implements the true chromecast receiver protocol:**
https://play.google.com/store/apps/details?id=com.softmedia.receiver.castapp

**These libraries also implement true chromecast receiver:**
https://github.com/enen92/script.tubecast
https://www.npmjs.com/package/yt-cast-receiver
https://www.npmjs.com/package/chromecast-device-emulator

hundredfire, 2024-04-24:
I'm "necro'ing" this topic, as I'm also very interested in this feature. This could allow us to make diy Chromecast audio very easily