iotsound/iotsound #103: Connection doesn't work.

zt3030, 2020-02-08:
Hi,
I've tried to install on 2 ver. 2.47+rev and 2.38+rev1.
On the beginning I've installed on 2.47 ver. and after parring bluetooth i had problems like "decoding error: No such file or directory".
I've read the other's issues, one user said that it works on 2.38 so I did the downdate to 2.38 and now I can't even connect bluetooth device. 
Also in both versions Airplay and Spotify connection doesn't work.

I'm using RPi 3b, and board jack audio exit.

AlexProgrammerDE, 2020-02-08:
@zt3030 could you give us some logs from the dashboard please?

zt3030, 2020-02-08:
I haven't saved logs from version 2.47+rev1. But here is log from attempt bluetooth connection on 2.48+rev1.:

08.02.20 19:09:45 (+0100)  bluetooth-audio  Setting output volume to 75%
08.02.20 19:09:45 (+0100)  bluetooth-audio  Connection notify volume is 75%
08.02.20 19:09:45 (+0100)  spotify  INFO:librespot: librespot (raspotify v0.13.1) 5d2cb32 (2019-04-16). Built on 2019-04-29. Build ID: bQrzTj3V
08.02.20 19:09:45 (+0100)  airplay  Device is discoverable as "balenaSound Airplay 6b20"
08.02.20 19:09:45 (+0100)  bluetooth-audio  amixer: Unable to find simple control 'Digital',0
08.02.20 19:09:45 (+0100)  bluetooth-audio  
08.02.20 19:09:45 (+0100)  bluetooth-audio  Restarting bluetooth service
08.02.20 19:09:50 (+0100)  bluetooth-audio  Bluetooth agent registered
08.02.20 19:09:51 (+0100)  bluetooth-audio  Starting bluetooth agent in Secure Simple Pairing Mode (SSPM) - No PIN code provided or invalid
08.02.20 19:09:55 (+0100)  bluetooth-audio  Device is discoverable as "balenaSound 6b20"
08.02.20 19:11:36 (+0100)  bluetooth-audio  AuthorizeService (/org/bluez/hci0/dev_84_C7_EA_AF_A8_1A, 0000110d-0000-1000-8000-00805f9b34fb)

After this I couldn't even find this device again. It just freezed. Restart hasn't helped

AlexProgrammerDE, 2020-02-08:
@zt3030 OK. You wrote too about a decoding error. Is it maybe related to the ones in #47?

zt3030, 2020-02-08:
> @zt3030 OK. You wrote too about a decoding error. Is it maybe related to the ones in #47?

Yes exactly it is. The solution for #47 was change the version to 2.38+rev1.
For me this downdate didn't helped.

zt3030, 2020-02-08:
Okay, i've pushed this version again and it just started working. Thx a lot for your help @AlexProgrammerDE

closed 2020-02-08 by zt3030