iotsound/iotsound #104: Problem with bluetooth paring when: "Attempting to reconnect (...)"

zt3030, 2020-02-09:
Hi,
What I discovered after one day of using Balena sound is, when I have log "Attempting to reconnect to previous bluetooth device: 84:**:EA:**:A8:**" I can't pair any device, even this which bluetooth adress is displayed (if it is android device) There is no such a problem with iPod.

tmigone, 2020-02-12:
Hi @zt3030, that log message indicates that device `84::EA::A8:**` was previously connected to balenaSound, so upon rebooting it will attempt to reconnect. Could it be that it is already connected? Im curious also if you redacted the MAC address or if it shows as `84::EA::A8:**`

tmigone, 2020-03-27:
Hey @zt3030 I'm closing this issue, if you still need help feel free to comment here. Cheers.

closed 2020-03-27 by tmigone