iotsound/iotsound #106: Services keep restarting

Cyclone47, 2020-02-12:
I have recently installed BelaneAudio. I notice the services keep restarting on my end. 

12.02.20 15:30:21 (+0100) Installed service 'spotify sha256:6b1bd29c46fe4bb756176bf313606601e424c89211f77ef9dae100d779c97fdb'
12.02.20 15:30:21 (+0100) Starting service 'spotify sha256:6b1bd29c46fe4bb756176bf313606601e424c89211f77ef9dae100d779c97fdb'
12.02.20 15:30:24 (+0100) Installed service 'bluetooth-audio sha256:6741f085a8a404e4e36b2304dfcf17d660f79ff9dc5059b8b799a9d8e6cd7961'
12.02.20 15:30:24 (+0100) Starting service 'bluetooth-audio sha256:6741f085a8a404e4e36b2304dfcf17d660f79ff9dc5059b8b799a9d8e6cd7961'
12.02.20 15:30:49 (+0100) Started service 'spotify sha256:6b1bd29c46fe4bb756176bf313606601e424c89211f77ef9dae100d779c97fdb'
12.02.20 15:30:41 (+0100)  spotify  INFO:librespot: librespot (raspotify v0.13.1) 5d2cb32 (2019-04-16). Built on 2019-04-29. Build ID: bQrzTj3V
12.02.20 15:30:41 (+0100)  spotify  INFO:librespot_core::session: Connecting to AP "gew1-accesspoint-a-890f.ap.spotify.com:4070"
12.02.20 15:30:42 (+0100)  spotify  INFO:librespot_core::session: Authenticated as "darkrogue" !
12.02.20 15:30:42 (+0100)  spotify  INFO:librespot_playback::audio_backend::alsa: Using alsa sink
12.02.20 15:30:42 (+0100)  spotify  INFO:librespot_core::session: Country: "NL"
12.02.20 15:30:57 (+0100) Started service 'airplay sha256:43ad390a718f42de63bbcba295dabf86937f68068e86af5070e99fd5f01c740e'
12.02.20 15:30:58 (+0100) Started service 'bluetooth-audio sha256:6741f085a8a404e4e36b2304dfcf17d660f79ff9dc5059b8b799a9d8e6cd7961'
12.02.20 15:30:50 (+0100)  airplay  Device is discoverable as "balenaSound Airplay 4160"
12.02.20 15:30:57 (+0100)  bluetooth-audio  Setting output volume to 75%
12.02.20 15:30:57 (+0100)  bluetooth-audio  Connection notify volume is 75%
12.02.20 15:30:57 (+0100)  bluetooth-audio  Restarting bluetooth service
**12.02.20 15:30:57 (+0100)  bluetooth-audio  amixer: Unable to find simple control 'Digital',0**
12.02.20 15:30:57 (+0100)  bluetooth-audio  
12.02.20 15:31:01 (+0100)  bluetooth-audio  Bluetooth agent registered
12.02.20 15:31:02 (+0100)  bluetooth-audio  Starting bluetooth agent in Secure Simple Pairing Mode (SSPM) - No PIN code provided or invalid
12.02.20 15:31:07 (+0100)  bluetooth-audio  Device is discoverable as "balenaSound 4160"
12.02.20 15:31:11 (+0100) Killing service 'bluetooth-audio sha256:6741f085a8a404e4e36b2304dfcf17d660f79ff9dc5059b8b799a9d8e6cd7961'
12.02.20 15:31:11 (+0100) Killing service 'spotify sha256:6b1bd29c46fe4bb756176bf313606601e424c89211f77ef9dae100d779c97fdb'
12.02.20 15:31:11 (+0100) Killing service 'airplay sha256:43ad390a718f42de63bbcba295dabf86937f68068e86af5070e99fd5f01c740e'

Any idea why this is happening?

tmigone, 2020-02-12:
Hi @Cyclone47 , what device and balenaOS are you running? From the logs the only thing worth noting is the line you highlighted, it seems that the `bluetooth-audio` service is unable to find an audio output. Are you running on a Pi 0 by any chance?

Cyclone47, 2020-02-12:
Hi @tmigone I am running on a Raspberry pi 3 with the latest balenaOS. 
I did manage to fix it by running "apt-get update && apt-get install --reinstall alsa-utils" as mentioned in https://github.com/balenalabs/balena-sound/issues/23. 

Not sure if this is something to look into though, as it didn't work on a clean install for me.
For me it's resolved, feel free to close if theres no further action/investigation needed.

jtonello, 2020-03-20:
I found that adding ```RUN install_packages alsa-utils``` to the ./spotify/Dockerfile.template resolved the playback issue. 

However, the existing ```start.sh``` script continued to fail after that change. A manual start of /usr/bin/librespot from the command line with simplified config parameters did work. I replaced the contents of ./spotify/start.sh with the same line and it worked:

``/usr/bin/librespot -u $SPOTIFY_LOGIN -p $SPOTIFY_PASSWORD -n $BLUETOOTH_DEVICE_NAME --enable-volume-normalisation --linear-volume --initial-volume=$SYSTEM_OUTPUT_VOLUME```

I'm running balenaOS 2.47.0+rev1 on an RPi 3 B+ with headphones plugged into the audio jack.

tmigone, 2020-03-27:
Hey @Cyclone47 what version of the Pi 3 were you using? I'm having a hard trouble reproducing this issue. @jtonello did you bump into this with a fresh install?

AlexProgrammerDE, 2020-05-02:
All fixes are included in the troubleshooting: https://landr-balenalabs-repo-balena-sound.netlify.app/docs/troubleshooting/ Closing.

closed 2020-05-02 by AlexProgrammerDE