iotsound/iotsound #107: Default system output volume is too low

chrisys, 2020-02-13:
I recently reduced the system output volume from a default of 100 to 75%, on the basis that if you're using some DACs, the output volume is far too loud and may cause damage to some systems.

As a result of that the standard Pi output is too low, and although it can be corrected by using the `SYSTEM_OUTPUT_VOLUME` env var, we should figure out some way to reduce the volume for affected DACs only and return the default to 100%.

AlexProgrammerDE, 2020-02-16:
@chrisys we can write a script which checks the RESIN_HOST_CONFIG. Do you know some devices which are affected?

chrisys, 2020-02-17:
@AlexProgrammerDE I'll have to check again to be sure but I think it was the `justboom-dac`.

cenix102, 2020-03-05:
@chrisys I had the same problem. If you are you using the balenaCloud just go to your device and add a "Device Variable".
name: SYSTEM_OUTPUT_VOLUME
value: 99

that all! enjoy your music
[image]

btw: with the variable BLUETOOTH_DEVICE_NAME you can change the device name for Bluethooth/Spotify (its the same variable)

pojashad, 2020-03-07:
I have a similar issue, I changed the ```SYSTEM_OUTPUT_VOLUE``` to 100 and I hear the change and volume gets loud for like 2 seconds and the connection drops and then the volume drops back to the standard value. It is the change doesn't get saved 
[image]

cenix102, 2020-03-07:
@pojashad  What do you use? Raspberry Pi 3+? or something else?

stef-1, 2020-03-07:
I have exactly the same issue as @pojashad, on a Raspberry Pi 3.

pojashad, 2020-03-09:
> @pojashad What do you use? Raspberry Pi 3+? or something else?

Raspberry Pi 4 Model B

tmigone, 2020-03-09:
hey @pojashad are you streaming via bluetooth?

chrisys, 2020-03-10:
Possibly related forum thread here: https://forums.balena.io/t/balena-sound-low-output/61927

@tmigone do you know if either of the snapcast containers exert any control over the volume?

tmigone, 2020-03-10:
I think it might be this bug: https://github.com/balenalabs/balena-sound/issues/78

pojashad, 2020-03-10:
> hey @pojashad @stef-1 are you streaming via bluetooth?

Yes I tried both bluetooth and airplay, same issue.

AlexProgrammerDE, 2020-04-10:
@chrisys what about setting it to 100 and warning the user that it can get extremly loud on a DAC?

tmigone, 2020-07-17:
Alex that's my initial approach too, but if the user misses the warning it might end up damaging their speakers, I would be mad :P
I think this could be solved with 3.x because of how pulseaudio deals with volumes. @chrisys do you remember what DAC you experienced this with? We can try and see if it's better.

njavilas2015, 2021-01-17:
this solution is...... 
the container balena-sound_audio and balena-sound_bluetooth have the volumen at 75%. that reduce at 50% total 
command for verifications is apk add alsa-utils alsa-utils-doc alsa-lib alsaconf and execute alsamixer and up volumen pleace excuse my english I'm noob

lesmo, 2021-01-25:
I've seen this happening when using the built-in sound output of the Raspberry. The `SYSTEM_OUTPUT_VOLUME` variable seems to be ignored (mine's set to 100), the `alsamixer` command shows 75%

tmigone, 2021-06-15:
This was addressed back then with 3.0 release. Note that `SYSTEM_OUTPUT_VOLUME` was deprecated in favour of `SOUND_VOLUME`.

closed 2021-06-15 by tmigone