iotsound/iotsound #111: Add support for IPS LCD on Pimoroni PirateAudio

barthex, 2020-02-27:
Pimoroni pHAT DAC has been replaced by the new Pirate Audio, which is equipped with a SPI LCD to display e.g. the images and name of played tracks. Pimoroni delivered own software based on Mopidy https://github.com/pimoroni/pirate-audio

tmigone, 2020-03-03:
Hi thanks for reporting this @barthex. We definitely want to add support for the Pirate Audio lineup but we haven't had the chance to play with one yet. If you get it to work please do share a PR!

barthex, 2020-03-03:
Hi, I'm the owner of Pirate Audio. It sounds beautiful, but in addition to the LCD, I have a haze how to set the right volume - the default causes painful clipping of the attached amplifier. Reducing the volume level through the Spotify Connect interface to a minimum is helpful.
I was thinking about using the -onevent option in librespot in combination with MQTT - this would allow external LCD, based on the popular ESP8266 with ESPEasy software. The doubt I have is whether to use an additional MQTT container, external MQTT (e.g. http://dioty.co). Both have their advantages.

Fiqq90, 2020-03-07:
Hi, I also got a Pirate Audio board. On top of the LCD screen, I was hoping that the 4 buttons on the headphone HAT could also be integrated with BalenaSound to skip/play/pause music over bluetooth/airplay/spotify. I saw that there is already a PR on adding bluetooth remote control, maybe this is something that can work in parallel?

newgas, 2020-09-04:
Hi @tmigone,
Could this project be compatible with balena sound?
https://github.com/pimoroni/pidi-spotify from https://github.com/dtcooper/raspotify/issues/294

Gadgetoid, 2020-09-10:
Is there anything I can do to help here?

@Fiqq90 In tinkering with Spotify and Raspotify/Librespot to build PiDi Spotify I found it's very much a one-way deal. Librespot's -onevent hooks can push information about play state and media into PiDi Spotify really easily, but the authentication procedure to exert control over a Spotify playlist (fully logging into the users account) is rather more complicated and I wasn't sure exactly where we stood in terms of a "commercial project".

Local volume control via alsa is possible, but may prove slightly contrived since if you turn the volume all the way down on the hardware you can't then turn it up from a remote client.

I think Bluetooth can support track information, is it possible to hook into this? It's a little outside the remit of PiDi-Spotify but the original PiDi - https://github.com/pimoroni/pidi (into which PiDi Spotify may ultimately be merged) - has both MusicBrainz album art lookup support, and the ability to - in theory - support client plugins to receive data from other sources.

I think ShairPort Sync can broadcast track metadata, too, which should be a shoo-in for PiDi.

Note: while Pirate Display is ostensibly for our products, it supports plugins that can - in theory - display album art on *any* display. I'm all ears for adding support for other LCD products if it better positions PiDi as the Get-Your-Track-Info-From-Whereever-Onto-An-LCD tool of choice.

saveriogzz, 2020-11-13:
Hi! I'm also the owner of a  Pirate Audio, how can I help here? @barthex

OldShaterhan, 2022-02-04:
Hi,

Is there any chance for progress on this issue?

Gadgetoid, 2022-02-07:
It's unlikely I'll ever find the time or headspace to look into support properly. I tend to be firefighting issues or working on new drivers most of the time and had completely forgotten about this.

I can potentially supply @tmigone with boards if they want a play, but otherwise I'm drowning in stuff to do.