iotsound/iotsound #112: bluetooth-audio and LG projector

jtonfat, 2020-03-01:
Hi!
I am trying to connect a raspberry pi zero W with a justboom-dac to a LG projector using bluetooth. 
However, the LG projector does not discover the raspberry pi. My smartphone is able to detect and connect through bluetooth to the raspberry pi but the projector does not. 
I have tested both bluetooth methods unsuccesfully : Secure Simple Pairing mode and Legacy mode. 
Also, I have tested setting a small device name.

Details about my setup:
Rpi Zero W + Justboom DAC Amp
HOST OS VERSION balenaOS 2.46.1+rev1
SUPERVISOR VERSION 10.6.27
TARGET RELEASE 8a93ce7

tmigone, 2020-03-27:
Hello @jtonfat, can your phone connect easily to the projector? Is the projector brand new or old? Do you have another Raspberry Pi device (maybe a Pi 3 or 4) to test? Perhaps you can scan for bluetooth devices from your Pi Zero to see if it can see the projector.

AlexProgrammerDE, 2020-04-10:
@jtonfat i will close this issue for now. We cant help you without more information. Feel free to reopen if you still have this issue.

closed 2020-04-10 by AlexProgrammerDE