iotsound/iotsound #113: Tinkerboard deployment fails

albydnc, 2020-03-08:
Hi everyone,
when I try to push the application to my Tinkerboard, I get
```
[Info]              Uploading images
[Success]           Successfully uploaded images
[Error]             Some services failed to build:
[Error]               Service: bluetooth-audio
[Error]                 Error: The command '/bin/sh -c install_packages   alsa-utils   bluealsa   bluez   mplayer   python-dbus   python-gobject   python-gpiozero' returned a non-zero code: 100
[Info]              Built on arm03
[Error]             Not deploying release.
Remote build failed

Additional information may be available by setting a DEBUG=1 environment
variable: "set DEBUG=1" on a Windows command prompt, "$env:DEBUG = 1" on
powershell, or "export DEBUG=1" on Linux or macOS.

If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io

For CLI bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/
```

AlexProgrammerDE, 2020-03-27:
@albydnc looks like not all needed dependencies for balena sound are available for the Tinkerboard. Can you please send the whole Log?

tmigone, 2020-03-27:
@albydnc as Alex said, there are most likely dependencies that are not available for the Tinkerboard. We haven't tested this project with that board. My first suggestion would be to remove `python-gpiozero` package from this [Dockerfile](https://github.com/balenalabs/balena-sound/blob/master/bluetooth-audio/Dockerfile.template#L13). It's not really essential for most functions and it's a package only available for Raspberry Pi family of boards.

AlexProgrammerDE, 2020-03-27:
@tmigone @albydnc if we figure this out can we create a specific Dockerfile, which only is used for the Tinkerboard. Something like Dockerfile.asus-tinker-board

AlexProgrammerDE, 2020-03-28:
@tmigone @albydnc i just checked the dependencies via a test project. There are multiple problems: bluealsa is not available via apt, python-gpiozero is not available via apt and snapcast works not with tinkerboard.
That is my full log:
$ balena push Test
[Info]              Starting build for Test, user g_alex_krem
[Info]              Dashboard link: https://dashboard.balena-cloud.com/apps/1622870/devices
[Info]              Building on arm03
[Info]              Pulling previous images for caching purposes...
[Success]           Successfully pulled cache images
[snapcast-server]   Step 1/10 : FROM balenalib/asus-tinker-board-debian:buster
[airplay]           Step 1/5 : FROM balenalib/asus-tinker-board-debian:buster
[bluetooth]         Step 1/17 : FROM balenalib/asus-tinker-board-debian:stretch
[snapcast-client]   Step 1/9 : FROM balenalib/asus-tinker-board-debian:buster
[spotify]           Step 1/4 : FROM balenalib/asus-tinker-board-debian:buster
[fleet-supervisor]  Step 1/7 : FROM balenalib/asus-tinker-board-alpine-node
[snapcast-server]    ---> f2e020951524
[snapcast-server]   Step 2/10 : ARG SNAPCAST_VERSION=0.17.1
[airplay]            ---> f2e020951524
[airplay]           Step 2/5 : ENV DBUS_SYSTEM_BUS_ADDRESS=unix:path=/host/run/dbus/system_bus_socket
[snapcast-client]    ---> f2e020951524
[snapcast-client]   Step 2/9 : ARG SNAPCAST_VERSION=0.17.1
[airplay]            ---> Running in 21bb8a5a4ab9
[fleet-supervisor]   ---> 5085b3059558
[fleet-supervisor]  Step 2/7 : WORKDIR /usr/src
[snapcast-server]    ---> Running in 33f3c8d70b61
[bluetooth]          ---> a84c21cf790b
[bluetooth]         Step 2/17 : ENV DBUS_SYSTEM_BUS_ADDRESS=unix:path=/host/run/dbus/system_bus_socket
[spotify]            ---> f2e020951524
[spotify]           Step 2/4 : RUN curl -sSL https://dtcooper.github.io/raspotify/key.asc | apt-key add -v - 2> /dev/null   && echo 'deb https://dtcooper.github.io/raspotify jessie main' | tee /etc/apt/sources.list.d/raspotify.list   && install_packages raspotify
[snapcast-client]    ---> Running in 06e72669e886
[bluetooth]          ---> Running in eaecf89f4ac5
[fleet-supervisor]   ---> Running in fbbbb1f76860
[spotify]            ---> Running in 1ba0695d2974
[airplay]           Removing intermediate container 21bb8a5a4ab9
[airplay]            ---> 7887b7d67fe3
[airplay]           Step 3/5 : RUN install_packages shairport-sync
[airplay]            ---> Running in b669286e952c
[snapcast-server]   Removing intermediate container 33f3c8d70b61
[snapcast-server]    ---> 9755b67f1aca
[snapcast-server]   Step 3/10 : WORKDIR /usr/src
[snapcast-client]   Removing intermediate container 06e72669e886
[snapcast-client]    ---> 0d594f113228
[snapcast-client]   Step 3/9 : WORKDIR /usr/src
[bluetooth]         Removing intermediate container eaecf89f4ac5
[bluetooth]          ---> 761622d7482b
[bluetooth]         Step 3/17 : ENV UDEV=1
[snapcast-server]    ---> Running in b738e473d1ca
[fleet-supervisor]  Removing intermediate container fbbbb1f76860
[fleet-supervisor]   ---> 58e7ae065016
[fleet-supervisor]  Step 3/7 : COPY . .
[snapcast-client]    ---> Running in ecc77c8b903c
[bluetooth]          ---> Running in add1ef1fd8d5
[snapcast-server]   Removing intermediate container b738e473d1ca
[snapcast-server]    ---> 7ec67a04ff76
[snapcast-server]   Step 4/10 : RUN install_packages  libavahi-client3  libavahi-common3  libflac8  libogg0  libopus0  libvorbis0a  libvorbisenc2  wget
[bluetooth]         Removing intermediate container add1ef1fd8d5
[bluetooth]          ---> 7a2e9b35485f
[bluetooth]         Step 4/17 : RUN install_packages   alsa-utils   bluealsa   bluez   mplayer   python-dbus   python-gobject   python-gpiozero
[snapcast-server]    ---> Running in 7b83c4eefc6c
[snapcast-client]   Removing intermediate container ecc77c8b903c
[snapcast-client]    ---> f26aa112e3bd
[snapcast-client]   Step 4/9 : RUN install_packages   libasound2   libavahi-client3   libavahi-common3   libflac8   libogg0   libopus0   libvorbis0a   wget
[fleet-supervisor]   ---> 3d98d536d0b8
[fleet-supervisor]  Step 4/7 : RUN JOBS=MAX npm install --only=production
[bluetooth]          ---> Running in 69653fb5f3a1
[snapcast-client]    ---> Running in 4c283dd272f2
[fleet-supervisor]   ---> Running in f34a49f98a1d
[spotify]           OK
[spotify]           deb https://dtcooper.github.io/raspotify jessie main
[airplay]           Reading package lists...
[fleet-supervisor]  npm WARN deprecated mkdirp@0.5.4: Legacy versions of mkdirp are no longer supported. Please update to mkdirp 1.x. (Note that the API surface has changed to use Promises in 1.x.)
[... 824 lines cut ...]
[snapcast-client]    ---> Running in 3c99a2ea1b89
[snapcast-server]   Removing intermediate container 0024fd5a54a0
[snapcast-server]    ---> 96bfce87297c
[snapcast-server]   Successfully built 96bfce87297c
[snapcast-client]   Removing intermediate container 3c99a2ea1b89
[snapcast-client]    ---> c5a3aef9211d
[snapcast-client]   Successfully built c5a3aef9211d
[Info]              Uploading images
[Success]           Successfully uploaded images
[Error]             Some services failed to build:
[Error]               Service: bluetooth
[Error]                 Error: The command '/bin/sh -c install_packages   alsa-utils   bluealsa   bluez   mplayer   python-dbus   python-gobject   python-gpiozero' returned a non-zero code: 100
[Info]              Built on arm03
[Error]             Not deploying release.
Remote build failed

Additional information may be available by setting a DEBUG=1 environment
variable: "set DEBUG=1" on a Windows command prompt, "$env:DEBUG = 1" on
powershell, or "export DEBUG=1" on Linux or macOS.

If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io

For CLI bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/

tmigone, 2020-03-30:
Thanks @AlexProgrammerDE! 
We would love to add support for the Tinkerboard in the future, for the time being as I don't have one it's very hard to do so. I'll see if I can get my hands on one but with the current covid situation I don't know when that could be. @albydnc if you make any progress feel free to share in this thread.

shawaj, 2020-12-27:
Just for reference, I get the following when using deploy with balena in the build log:

```
[Error]             Some services failed to build:
[Error]               Service: audio
[Error]                 Error: manifest for balenablocks/audio:asus-tinker-board not found: manifest unknown: manifest unknown
[Error]               Service: bluetooth
[Error]                 Error: manifest for balenablocks/bluetooth:asus-tinker-board not found: manifest unknown: manifest unknown
[Info]              Built on arm01
[Error]             Not deploying release.
```

The full log outputs are as follows...
* [audio-log.txt](https://github.com/balenalabs/balena-sound/files/5744009/audio-log.txt) - fails
* [bluetooth-log.txt](https://github.com/balenalabs/balena-sound/files/5744010/bluetooth-log.txt) - success
* [balenasound-log.txt](https://github.com/balenalabs/balena-sound/files/5744011/balenasound-log.txt) - fails

shawaj, 2020-12-27:
@tmigone I guess this is because there are no images on docker hub balenablocks repo? Due to them not being in the build-images scripts?

https://github.com/balenablocks/bluetooth/blob/master/scripts/build-images.sh
https://github.com/balenablocks/audio/blob/master/scripts/build-images.sh

Would it be sufficient to just add the following to the build scripts? Or is there more work that needs to be done?
```
build_and_push_image $DOCKER_REPO "asus-tinker-board" "linux/arm/v7"
build_and_push_image $DOCKER_REPO "asus-tinker-board-s" "linux/arm/v7"
```

shawaj, 2021-01-01:
Added some PRs for those:

Bluetooth - https://github.com/balenablocks/bluetooth/pull/8
Audio - https://github.com/balenablocks/audio/pull/50

tmigone, 2021-02-09:
@shawaj were you able to deploy balenaSound to the tinkerboard? I reckon as long as you remove the spotify plugin (it relies on raspotify which is raspberry pi specific) it should work.