iotsound/iotsound #115: AirPlay is not visible/working

volisok, 2020-03-10:
Hi! 
I am new to balena sound, but I like it already. I have balenaOS 2.47.0+rev1 installed on Pi3 with recent balena sound master. BT is working good (after adding the SYSTEM_OUTPUT_VOLUME). The AirPlay service is running, but not visible on Apple devices. What could be wrong?

PS Is there a Device Environment Variable for disabling SPOTIFY service? I do not use it at all and wish to stop it. 
Is there a list of all available DE variables?

BR

volisok, 2020-03-16:
no other services are running on the Pi3. bOS and balena sound were installed using your step by step manual.

tmigone, 2020-03-27:
Hi @volisok, 

> BT is starting with level 67 (even after adding the SYSTEM_OUTPUT_VOLUME 90 to DEV). The AirPlay service is running (in Dashboard), but not visible for Apple devices. What could be wrong?

Are you setting `CONNECTION_NOTIFY_VOLUME`? You might be hitting this bug: https://github.com/balenalabs/balena-sound/issues/78

> PS Is there a Device Environment Variable for disabling SPOTIFY service? I do not use it at all and wish to stop it.

There is no env var to disable that service but you can easily do so by removing (or commenting out) spotify service definition from the `docker-compose.yml` file and re-deploying your app. This is what you need to remove:
```
  spotify:
    build: ./spotify
    restart: always
    network_mode: host
    privileged: true
    volumes:
      - spotifycache:/var/cache/raspotify
      - snapcast:/var/cache/snapcast
```

> Is there a list of all available DE variables?

We don't have a compiled list, but all of them are listed on the README file: https://github.com/balenalabs/balena-sound/blob/master/README.md
We are working on better displaying and organizing this information.

cflachen, 2020-04-09:
Hi everybody,
I had the same problem with Air Play on my iPhone not finding my Raspi3. Since I only have 1 system running balena-Sound, I set the environment variable DISABLE_MULTI_ROOM to 1, and now the Raspi is recognized by my iPhone.
I hope this will help in debugging the problem.
Cheers, Claus

tmigone, 2020-07-17:
Closing as it seems to be sorted out. Feel free to reopen.

closed 2020-07-17 by tmigone