iotsound/iotsound #117: multi-room: volume automatically dropping to 36

tmigone, 2020-03-11:
Raspberry Pi 4 with balenaOS 2.47.1+rev1. And that´s correct I´m NOT using DAC.

MatthewCroughan, 2020-03-27:
I can vouch for this. I'm using balenaOS v2.38.0+rev1, as this is required due to #62 

Despite setting the device variable `SYSTEM_OUTPUT_VOLUME` to `100` I inspected the `bluetooth-audio` container and ran `alsamixer`. I found that the volume was 36, same as the issue title. I turned it up manually here.

MatthewCroughan, 2020-03-27:
Ah! I think I've observed the behavior fully. Here's what occurs.

- Device boots
- Variable is applied somewhere in the entrypoint
- Bluetooth connection is established
- Music plays at 100% volume initially for around 1-2 seconds
- Volume reverts to 36 seconds after bluetooth connection is established

MatthewCroughan, 2020-03-27:
Indeed, every time a bluetooth device is connected, the volume reverts to 36.

`[  847.092383] input: 28:16:7F:63:61:91 as /devices/virtual/input/input2` this also occurs and is visible in the kernel log at the same time.

Interestingly, it seems udevd is having a hard time with these virtual inputs, and sometimes it actually kills them. At one point this did actually cause the volume to drop back down to 36, but the audio still sounded as if it was intact. In other words, all I perceived audibly was a volume drop back down to 36.

```
[  504.409162] systemd-udevd[29]: seq 1854 '/devices/virtual/input/input0' is taking a long time
[  544.409371] systemd-udevd[29]: seq 1861 '/devices/virtual/input/input1' is taking a long time
[  624.409862] systemd-udevd[29]: seq 1854 '/devices/virtual/input/input0' killed
[  624.411246] systemd-udevd[29]: worker [107] terminated by signal 9 (Killed)
[  624.411290] systemd-udevd[29]: worker [107] failed while handling '/devices/virtual/input/input0'
[  664.410022] systemd-udevd[29]: seq 1861 '/devices/virtual/input/input1' killed
[  664.411363] systemd-udevd[29]: worker [106] terminated by signal 9 (Killed)
[  664.411393] systemd-udevd[29]: worker [106] failed while handling '/devices/virtual/input/input1'
[  685.410126] systemd-udevd[29]: seq 1858 '/devices/virtual/input/input0' is taking a long time
[  805.410831] systemd-udevd[29]: seq 1858 '/devices/virtual/input/input0' killed
[  805.413128] systemd-udevd[29]: worker [105] terminated by signal 9 (Killed)
[  805.413169] systemd-udevd[29]: worker [105] failed while handling '/devices/virtual/input/input0'
[  847.092383] input: 28:16:7F:63:61:91 as /devices/virtual/input/input2
[  896.411215] systemd-udevd[29]: seq 1864 '/devices/virtual/input/input1' is taking a long time
[  907.411279] systemd-udevd[29]: seq 1867 '/devices/virtual/input/input2' is taking a long time
```

tmigone, 2020-03-30:
Thanks @MatthewCroughan for the thorough investigation. I'll work on fixing this shortly, the information you compiled surely does help.

MatthewCroughan, 2020-04-04:
I've just solved this issue. It might be a good idea to change the title of this issue since it doesn't seem to be a problem with multi-room specifically, but rather with the use of the `amixer sset` commands in the udev rule and entrypoint of the `bluetooth-audio` container.

https://github.com/balenalabs/balena-sound/blob/ce2470e07bf74a6fa4b96271b6ee9056d78898a8/bluetooth-audio/start.sh#L21-L22

https://github.com/balenalabs/balena-sound/blob/ce2470e07bf74a6fa4b96271b6ee9056d78898a8/bluetooth-audio/bluetooth-udev#L22-L23

Let's say I want to use 50% volume in the container, I'll execute `amixer sset PCM,0 50%`

The problem is that by default, amixer is not using the command line option -M which means "Mapped Volume", and instead is being supplied a 'raw value' which does not equate to the human readable/audible volume mapping we want. For example, without -M, `50%` is equal to `13%` as can be seen in the AlsaMixer gui, or a value of -4919 which is close to -50dB. I have no idea what is actually happening here since I'm not clued up on these systems. The absence of a % sign also changes this behaviour, so supplying `100%` vs `100` changes the value.

## Before -M (mapped volume)
[image]

## After -M (mapped volume)
[image]

This may have been overlooked because the value 100% is the max number, no matter the supplied options, so someone may have tested with 100% and assumed it worked since it changes from the default value.

# TL;DR
Simply supplying -M to the `amixer` commands fixes this.

This information was gleamed from [this stackoverflow post](https://superuser.com/questions/1469453/difference-between-alsamixer-and-amixers-output-about-the-percentage-of-master)

I'll submit a pull request with the fix shortly.

MatthewCroughan, 2020-04-04:
Excuse the poor commits and force commits, I was just confused about how to actually follow the contribution guidelines. I did not know that putting multiple commit messages would separate them out into paragraphs like that.

closed 2020-04-08 by tmigone