iotsound/iotsound #118: Bluealsa - No such file or directory

BlackFlagTV, 2020-03-17:
Hello,

I'm using an Raspberry Pi 3 with balenaSound. When I hear about Airplay, for example, everything sounds fine, but when I hear through Bluetooth, it sounds strange and I get an error message in the console:
`17.03.20 19:05:55 (+0100)  bluetooth-audio  /usr/bin/bluealsa: SBC decoding error: No such file or directory
17.03.20 19:05:55 (+0100)  bluetooth-audio  /usr/bin/bluealsa: Missing RTP packet: 585 != 583
17.03.20 19:05:55 (+0100)  bluetooth-audio  /usr/bin/bluealsa: SBC decoding error: No such file or directory`

tmigone, 2020-03-27:
Hello @BlackFlagTV, can you post a larger extract of your logs? I'm mainly interested in knowing if the `Missing RTP packet` line you posted is being logged repeatedly. If that is the case, it's an indicator that you are hitting the issue described here: #62, specifically of interest [this](https://github.com/balenalabs/balena-sound/issues/62#issuecomment-576307747) post.

AlexProgrammerDE, 2020-04-29:
No activity. Closing.

closed 2020-04-29 by AlexProgrammerDE